arkypita/LaserGRBL #175: Laser GRBL end Arduino mega + ramps 1.4

Pegaso72, 2017-11-19:
I'm sure everyone is Sandro from Italy,
I wanted to make a Laser cnc pivvolo with a 6 watt power diode.
I wanted to use the same electronics I use my 3D printer.
I use Marlin Kimbra in the arduino card but as I open the Laser GRBL program does not connect (115200 communication port and baudrate)
What can I do ?

arkypita, 2017-11-19:
Marlin is not Grbl
LaserGRBL expects to talk with Grbl (v0.8, v0.9 or 1.1) and it requires to receive responses (such as the status message and position) in the Grbl format, otherwise it can not validate the connection.

arkypita, 2017-11-19:
Realtime status is reported in this format by grbl
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000> (v1.1)
<Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000> (v0.9)

LaserGRBL expect to receive this kind of message from grbl otherwise it remains connecting/disconnected.

Pegaso72, 2017-11-19:
Ok grazie :-(

closed 2017-11-21 by arkypita