arkypita/LaserGRBL #177: Uneven burning

Mikalay31, 2017-11-23:
Hello ! I have such a problem. The edges near the picture are burned deeper. I understand this is due to the fact that in these places the direction of motion changes. Acceleration costs a little (300). And so the laser burns more. Whether there is such function that around of a picture there was a virtual field of white color. To change the direction does not coincide with the moment of burning. Thank you ! Nicholas.

Mikalay31, 2017-11-23:
http://prntscr.com/heesdn            http://prntscr.com/heesov

arkypita, 2017-11-24:
Use Grbl v1.1 with laser mode enabled and M4 Dynamic Laser Power Mode to get the best result.
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode#laser-mode-operation

arkypita, 2017-11-24:
[image]

[image]

arkypita, 2017-11-24:
Also the suggested/required feature is already in roadmap #64 (acceleration offset)

closed 2017-11-24 by arkypita

Mikalay31, 2017-11-27:
Thank you ! That helped ! Thank you for the excellent program. As always, I follow your updates.