arkypita/LaserGRBL #180: Multipale pass problem

Umnik-va, 2017-11-25:
Hi!

Very nice programm, thank you very match!!

Since version 2.8.18 the function of multiple passes does not work. On version 2.8.16, everything works fine if you specify several passes and click on the "Start" icon - the program executes the selected number of passes, after each pass the counter is reduced by one. Starting with version 2.8.18 when I click on the "Start" icon the program does only one pass, the pass count is reduced by one and the code is not re-executed. To start the pass one more time, you have to blink again by the start icon.

Начиная с версии 2.8.18 не работает функция множественных проходов. На версии 2.8.16 все работает нормально, если указать несколько проходов и кликнуть по значку "Старт" - программа выполняет выбранное количество проходов, после каждого прохода счетчик уменьшается на единицу. Начиная с версии 2.8.18 когда я нажимаю на значок "Старт" программа делает только один проход, счетчик проходов уменьшается на единицу и повторно код не выполняется. Для запуска прохода еще один раз надо снова клинкуть по значку старт.

Vladimir

StuartB4, 2017-11-26:
Same here in V 2.8.20

arkypita, 2017-11-26:
Thanks for report. I check as soon as possible

closed 2017-11-26 by arkypita

arkypita, 2017-11-27:
fixed in v2.8.22
was a side effect of some change I made for "job resume feature" in v2.8.16

thanks for feedback

commit 5c328998 (2017-11-26): fix #180