arkypita/LaserGRBL #181: How to turn off spindle in pause mode?

Evil-Nemesis, 2017-11-26:
hello there
I need to turn off the spindle when I pressed feed hold button!
how can I do it!?

arkypita, 2017-11-26:
This behaviour is managed by grbl, not by LaserGRBL

arkypita, 2017-11-26:
use v1.1 with laser mode to obtain required behaviour
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode

closed 2017-11-26 by arkypita

Evil-Nemesis, 2017-11-28:
hi arkyoita.
**first of all I really thank you for your efforts!**
and thanks a lot for your answer!
I actually do that!!
for hardware button I do this:
uncomment this line in config.h:
#define ENABLE_SAFETY_DOOR_INPUT_PIN

and for the software button I changed this:
`void BtnStopClick(object sender, EventArgs e) { Core.FeedHold(); }`
to this
`void BtnStopClick(object sender, EventArgs e) { Core.FeedHold(); Core.SafetyDoor(); }`
and it's OK!