arkypita/LaserGRBL #183: Custom buttons

vaicekas, 2017-11-27:
Hello
Thank you very much for the great program!
But I would really like a feature that would allow to enlarge the Trace frame at the custom distance. It would be used for cutting according to the contour of the picture.
I imagine this as using GCode shortcuts and mathematical steps. For example: G1 X [left] -5 Y [botom] -5
G1 X [right] +5 Y [botom] +5 and so on.
If this were not tough, we would be very happy with this feature in future versions of the program.
Thank you.

arkypita, 2017-11-27:
It's something I always thought someone would one day ask for.

It is not too difficult, but it requires to implement of a kind of interpretation of the text a little more complicated than "search for [left] and replace with a number"

vaicekas, 2017-11-27:
it means You are missing it to :) 
So it can be expected that this will happen sometime ;)

arkypita, 2017-12-14:
try [v2.8.31](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.31)

closed 2017-12-14 by arkypita