arkypita/LaserGRBL #184: M3 / M4

B1Gt0ne, 2017-11-27:
The option to use M4 in the drop down when creating gcode box has disappeared. It only shows M3

arkypita, 2017-11-28:
Yes, M4 is only available if you have grbl 1.1 with "laser mode" enabled ($32=1) otherwise M4 is not effective (i think).

So, enable laser mode using "grbl configuration" form, and you will have M4 option appare.

B1Gt0ne, 2017-11-28:
I use LaserGRBL as a 'stand alone' program. I then save the gcode to a memory stick to take to my machine.
Until the latest updates I was able to choose M3 or M4 . 
I have also noticed that the M5 dropdown is also stuck on M5 and the jog speed option is missing

arkypita, 2017-11-28:
> I use LaserGRBL as a 'stand alone' program. I then save the gcode to a memory stick to take to my machine. Until the latest updates I was able to choose M3 or M4 .

Oh... i am sorry! In the latest versions I have connected the features available in LaserGRBL with Grbl configuration readed from arduino. This is to prevent the usage of settings that are not supported by HW (i.e. no homing button if homing is disabled etc...).

I did not think about the possibility of stand-alone usage as gcode generator :-(

I think I will release a new version that open all the features by default, if the grbl configuration is unknown by LaserGRBL.

Meanwhile If you need to have M3/M4 option you can simply connect LaserGRBL to your arduino "one time" and read the configuration. LaserGRBL will store the configuration permanently and allow you to use all the feature of your engraver. Is it possible for you?

> I have also noticed that the M5 dropdown is also stuck on M5 and the jog speed option is missing

M5 is the only option for laser off, so it is ok.

Jog speed option was taken by LaserGRBL in older version but was totally useless because grbl ignore the speed in G0 commands. G0 is always executed at the maximum speed settled in $110, $111
So, I removed job speed in raster import procedure

arkypita, 2017-11-28:
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.23

closed 2017-11-28 by arkypita

B1Gt0ne, 2017-11-28:
Many thanks for your response arkypita. All sorted now.

Excellent program. Keep up the good work

commit f315f9e9 (2017-11-28): fix #184