arkypita/LaserGRBL #186: engraved color!

Evil-Nemesis, 2017-11-28:
hello there.
I thank you again and again for this perfect project.

I want to change the color of the places that are engraved!
In other words, draw something on preview bitmap while the machine is engraving.
How can I do that?

arkypita, 2017-11-28:
LaserGRBL use GDI+ to draw preview. GDI+ performance is not so fast to allow the redraw of a complex image in small times, so it is not so easy to perform a real-time drawing of the progress.
https://github.com/arkypita/LaserGRBL/issues/158

However, if you want to try something, feel free to do your changes on file GrblPanel.cs

The function `protected override void OnPaint(PaintEventArgs e)` is called periodically and can be used to compose the image. Currently the OnPaint compose an image with a cached bitmap with the preview and the cursor position positioned over it

dungjk, 2017-11-29:
Have you try following libraries
https://github.com/luca-piccioni/OpenGL.Net
https://github.com/luca-piccioni/OpenGL.Net/tree/master/OpenGL.Net.Objects (If you want OOP)

arkypita, 2017-12-16:
Thanks for the hint! Since it is not a priority I add to dev roadmap but cannot know when and if could be developed :-)

closed 2017-12-16 by arkypita

urkmandeer, 2020-05-02:
I started implementing a couple of features for personal use. Adding this was simple enough but since I'm not a C# dev it's most likely not up to anyone's standards.

It only redraws what is needed, so nice and quick. 

The 3.6MB file below (220k lines) updates in realtime. The original code takes about 4 seconds for a full redraw.

[image]

Commands sent (pending) in yellow, commands completed in green.

[image]

arkypita, 2020-05-02:
Awesome!
If you are want to share your work with the community feel free to let me know when you think it is stable and "production ready". If you are able you can send changes with a github push, or sending source code to me at info@lasergrbl.com

I will be happy to examine it and insert it into the program

urkmandeer, 2020-05-03:
Sure thing, not problem sharing. 

The code won't be up to standard, so I pushed the related changes into a fork. The commit can be found [here](https://github.com/urkmandeer/LaserGRBL/commit/8be50889bfb8ecf805eda2c35ce48ff0bbf7ce80).

In short, I added a command number to GrblCommand, then keeping track of that number for the last queued and ack'ed commands. This is used determine what color should be used when drawing the graphic.

arkypita, 2020-06-02:
Hi @irkmandeer 
I experimented with your code but I ran into several problems including
- increased CPU usage
- memory leak (cumulative effect of redraw that does not free memory)
- misalignment on restarting work, in the face of interruptions or multiple passes

I tried to rewrite your code according to a different approach, instead of using the index in the list I proceeded to generate a numeric "id" code of each gcode line in order to recognize the point where it arrived etc but I still had problems therefore I temporarily gave up this job.