arkypita/LaserGRBL #187: Laser engraving with arduino mega 2560 and cnc shield ramps v1.4 and 3 stepper motors

LifeIsMySchool, 2017-11-28:
Hi friend! Can you help me, please? Can I use your grlb software and laser grlb editor at arduino mega 2560 and cnc shield ramps v1.4 and 3 stepper motor? Is it possible? Are they appropriate for my component?﻿ My design will be like at the photo.
[image]

arkypita, 2017-11-29:
Grbl is not my product. The github page for GRBL for mega is https://github.com/gnea/grbl-Mega/
If you use GRBL - LaserGRBL can work with your board.

closed 2017-11-29 by arkypita

LifeIsMySchool, 2017-11-29:
Thank you. Is this grbl for laser engraving? How can I upload at the mega? Because Arduino software editor warned me that it accepts .ino or .pde extension. How can I also use CNCShield Ramps V1.4? Do you know what these files are?
[grbl-master.zip](https://github.com/arkypita/LaserGRBL/files/1513697/grbl-master.zip)
[ArduinoMegaPololuShield.zip](https://github.com/arkypita/LaserGRBL/files/1513698/ArduinoMegaPololuShield.zip)
[UltiMachineRAMPS1-4Sprinter.zip](https://github.com/arkypita/LaserGRBL/files/1513699/UltiMachineRAMPS1-4Sprinter.zip)

LifeIsMySchool, 2017-11-29:
I installed Grbl Desktop before two days.

LifeIsMySchool, 2017-11-29:
I cloned mega in the githup desktop. But I don't now what it means. Because I can't add any grbl file from my computer?

arkypita, 2017-11-29:
https://github.com/gnea/grbl/wiki/Compiling-Grbl
https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino

LifeIsMySchool, 2017-12-06:
Do you have Arduino cod for two stepper motor, one laser head and one fan? I really need it.