arkypita/LaserGRBL #189: Setting request

Andy-Gadget, 2017-11-30:
Great software, but could I request a feature :
When engraving on a light coloured material (e.g. birch wood) the laser sometimes does not char quickly enough to start a track so the line isn't engraved.  Another GCode generator (I think it's LaserWeb) has a 'delay' setting, which, after a laser off move, energises the laser then waits for a time before starting the engraving move.  Sensible default for this would be 0 (i.e. no delay) up to 200 ms or so.
Andy.

arkypita, 2017-12-01:
I know the problem, is very common with light coloured materials and low power lasers.
Do you know how LaserWeb create delays? I mean, what is the GCode used? G4?

Andy-Gadget, 2017-12-02:
It wasn't LaserWeb; It's the JTech plugin for Inkscape which has this setting, and looking at the generated code it is as you suggest, using G4 for the delay .  In the code below, G4 P1.23... is a 1.23 second delay I set in the configuration. (P sets the delay units to seconds.)
Andy.
```
. . .
G1  X13.97 Y32.0322
G4 P0 
M03 S255
G4 P1.2300000190734863
G1 F250.000000
G1  X29.7744 Y12.5589
G4 P0 
M05 S0
G1 F500
G1 X0 Y0
M18
```

arkypita, 2017-12-14:
added to roadmap

closed 2017-12-14 by arkypita

BitterAndReal, 2018-05-20:
I'm really looking forwad to this feature. When do you expect this?

arkypita, 2018-05-22:
a lot of feature request and few spare time now... i cannot tell when I can do this

FrankHovis, 2021-03-07:
Has this been implemented or not? I see a few issues opened with the same problem, the laser takes a little while to switch on, but I can't see as there has been any actual way to resolve this using laserGrbl.

sully9, 2023-06-20:
Apologies for the chatter, but I agree this feature would be useful.  In the meantime, a tedious, manual workaround I use is: 

1. Load PNG/SVG into LaserGRBL
2. Save resulting Gcode to .NC file
3. edit .NC with text editor, add a 'G4 P0.025' command following any spindle(laser) ON command ('Sxxx') 
4. Load the .NC file into LaserGRBL

One of my everyday setups is a very cheap "DIY CNC2418" with a similarly cheap 0.5W laser.  The laser is very slow to respond to spindle commands,  So if I configure  acceleration and feedrates for a tolerable run time, there's a lot of incomplete lines - even on paper.  So once I've sorted out all the other relevant adjustments, I use the steps above.