arkypita/LaserGRBL #190: M7 M8 M9 commands

dixib, 2017-11-30:
hi arkipita 
i think that you did a great work with this software for peoples like me (mean hobbysts :-)..)
and ask if you can add a feature that can allow activate flood and/or mist  with relative M words
when g-code is generated from an image 
tnx alot for your work

arkypita, 2017-12-01:
I plan to review much of the GCode generation-optimization algorithm to make it more flexible and configurable. In a not too far future I will also add this feature.

For now i add it to the development roadmap #64

closed 2017-12-01 by arkypita

arkypita, 2019-11-23:
You can test this version: https://github.com/arkypita/LaserGRBL/releases/tag/v3.0.19
Maybe custom header/footer can be a solution for you.