arkypita/LaserGRBL #191: Change M3 / M4 to M106 gcode

agasevangelos, 2017-11-30:
Hi is there a way to generate gcode with custom ON/OFF M-code for the laser?
I would like to use this gcode on a 3dprinter running Marlin.

Thanks

arkypita, 2017-12-01:
I plan to review much of the GCode generation-optimization algorithm to make it more flexible and configurable (i.e. change the kind of command for laser on/off). In a not too far future I will also add this feature.

As workaround you can save the file and edit it as a text file (find-replace).

closed 2017-12-01 by arkypita

bywebberen, 2017-12-06:
I tried that, renaming it to *.gcode from *.nc with all M3 inside replaced with M106 And M5 replaced with M107 still no luck...

arkypita, 2017-12-06:
What do you mean with no luck? What is the problem?

bywebberen, 2017-12-06:
The the program runs(the steppers) , but it doesn't fire the laser on and
off.

Venlig hilsen / best regards
Niels Østbjerg
Syrenvænget 8
5792 Årslev

+45 40630201

bywebberen, 2017-12-07:
Did the issue get resolved by anyone out there?