arkypita/LaserGRBL #192: incomplete cut

beytullahgol, 2017-12-01:
as seen in the picture, the cut does not end

[image]
[image]
[image]

arkypita, 2017-12-01:
Is this an issue of LaserGRBL?

beytullahgol, 2017-12-01:
i don't know.. maybe you can help or fix this with software editing

arkypita, 2017-12-01:
Maybe laser turn on but material start melting/cutting only a little bit later OR Laser is commanded to turn on by Grbl but electronic circuits add some delay in switch on.

I think this problem/solution is similar to https://github.com/arkypita/LaserGRBL/issues/189

beytullahgol, 2017-12-01:
machine I used CTK-1350C Cutting Plotter 
electronic circuits can create delays
the delay will work

closed 2017-12-14 by arkypita