arkypita/LaserGRBL #193: new issue: Show the Filenmae of the loaded File

midibase, 2017-12-02:
Hello,
as the title said: 
It would be nice if I could see the filename from the loaded file e.g. in the title

best regards

sigi

arkypita, 2017-12-03:
[image]

closed 2017-12-03 by arkypita

midibase, 2017-12-03:
Hello, thank you for your replay.

I know that, but the problem is, if you have your files saved on a network (as I 😊 ), the Fullfilename can be too long

for the textbox. So you can’t see at a glance the filename.

This is for me importand because I my files named

Wald1.jpg, Wald2.jpg etc.

 

Best regards

 

Sigi

 

 <https://user-images.githubusercontent.com/8782035/33527454-af1126d8-d851-11e7-844e-f1f3b8383fb3.png>

arkypita, 2017-12-03:
I don't like to have the filename in form title.
I think i will put the filename only in the texbox and the full path as tooltip on it

arkypita, 2017-12-03:
done in v2.8.26
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.26

midibase, 2017-12-03:
Thanks,

that‘s a good solution

 

best regards sigi