arkypita/LaserGRBL #195: Problem resuming job

hk89, 2017-12-03:
I usually start from a position away from the machine zero and set a new work/home position. When an error happens, I reset the arduino. After that I run the gcode and I choose to do a resume from the last line sent.  It does not resume a job from my stored work/home position. What happens is that it goes to the machine zero X0 Y0 correctly and then goes to the last stopped line in absolute machine position and resumes from there as if the machine and work positions are both at X0 Y0. That means that it ignores the saved work position that I started with.

arkypita, 2017-12-03:
Hi @hk89 
Job resume does a fantastic job, however it can not do miracles!

The **working offset** that you assign when you set your new home position is stored and managed in grbl (not in LaserGRBL, that only show it as a second pair of coordinates) so it is lost if you perform a SOFT RESET (ie. using thunder button). You can see that the offset is lost because the second pair of coordinate disappear.

Since "Job Resume" system work only using the info available in gcode file, all the "external" commands that could have changed the world around it should be restored manually.

I mean: if you have moved (jog) to 100,100 and you have set a new home prior to start the job, you should re-position to 100, 100 and set home again prior to resume job.

I think that something of this process could be managed by LaserGRBL, by keep track of the previous working offset and try to re-apply the offset if detect grbl was reset and offset is lost.

Furthermore: if you do an HARD RESET (i.e. pushing the reset button of your board - or detaching power) you loose al the position information, so the only way to recover from this situation is to have limit-switch and perform homing to sense an absolute and reliable "0,0" point.

closed 2017-12-03 by arkypita

arkypita, 2017-12-03:
Try v2.8.26 and let me know
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.26

hk89, 2017-12-04:
Hi,
Thanks for the fast reply. I tried it twice and unfortunately it did not work. I guess I am missing something.

1. loaded up a vector image and generated gcode
2. homed my machine using limit switches
3. jogged to X70 Y20 as an example and set a work position there
4. ran gcode program and pressed hold after sending 422 lines
5. reset grbl from LaserGRBL
6. resumed job using the last command sent (option 3), and choose to redo homing and use the home position X70 Y20

Then it started homing and after that the machine started engraving from X0 Y0, although I can see the G92 command sent. It then triggered my limit switches and stopped.

arkypita, 2017-12-04:
Try without redo homing at step 6

hk89, 2017-12-04:
Yep! That worked! Thank you

I haven't got any idea why it did not work with redoing homing.

hk89, 2017-12-04:
I guess I know what is happening. After redoing homing it sends the G92 command at the homed machine coordinates X0 Y0. I guess what it should be like is to:

1. Redo homing
2. Jog to the work position known by LaserGRBL
3. Set the G92 command from there with the X and Y values
4. Resume from last line sent

Maybe it is missing the jogging part at step 2.

arkypita, 2017-12-04:
Yes something similar.

I have implemented a restore WCO (work coordinate offset) routine that require no other changes or movement should be done - included homing. It is implemented like this:

```
CurPoint = MachinePosition (read from actual) - LastValidWCO (stored when executing job)
G92 CurPointX CurPointY
```

This is equivalent to say "the point you are now placed has coordinates of X,Y"
For example: if prior to hang we have a WCO of 100,100 and job is at 10,10 when we reset grbl the absolute (machine position) is 110,110.

Compute MachinePosition (110,110) - LastValidWCO (100,100) give the result of 10,10
Sending G92 X10 Y10 is enough to restore the WCO at 100,100 without need jogging.

This is beacause G92 could be called not only with X0 Y0 but also with any XY position, with the meaning of say "the current place should be now known as X,Y"

However this trick does not work if you perform homing because the CurPoint coordinate are computed prior to send $H so they are computed as 10,10 (when they must be computed as -100, -100)

arkypita, 2017-12-04:
I have now changed it to 

```
if (setwco)
{
	//compute current point and set offset
	System.Drawing.PointF pos = homing ? new System.Drawing.PointF(0, 0) : MachinePosition;
	System.Drawing.PointF wco = mTP.LastKnownWCO;
	System.Drawing.PointF cur = new System.Drawing.PointF(pos.X - wco.X, pos.Y - wco.Y);
	mQueue.Enqueue(new GrblCommand(String.Format("G92 X{0} Y{1}", cur.X.ToString(System.Globalization.CultureInfo.InvariantCulture), cur.Y.ToString(System.Globalization.CultureInfo.InvariantCulture))));
}
```

The line `System.Drawing.PointF pos = homing ? new System.Drawing.PointF(0, 0) : MachinePosition;` say: _"if homing require compute position from 0,0 else from current position"_

This should work. (i have a doubt if after homing the position is 0,0 for definition, or should be something different in grbl)

arkypita, 2017-12-04:
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.27

hk89, 2017-12-05:
That's an interesting issue and outstandingly fast reply from you @arkypita. Keep up the good work.

hk89, 2017-12-17:
Hi,

Sorry to bump again but the latest release just did not have the option to resume from a home position. I did not see the checkbox. Yesterday, I ran into a problem and just couldn't resume. Please help. 

Thanks

arkypita, 2017-12-18:
I guess you are talking about Homing $H. Since v2.8.16 - v2.8.20 LaserGRBL use the configuration readed from grbl ($$) to self-configure some feature.

One of those feature is homing. LaserGRBL read $$ response and detect value of $22 homing enabled (or $17 for older version of grbl) and enable the homing buttons of its UI only if homing is enabled.
This affect also the ability to do a homing from resume interface.

The configuration is readed by LaserGRBL only when you configure from the new config interface, if you configure sending $NUM=VAL from command line LaserGRBL does not intercept it.

[image]

If you have the Homing button in your UI you should have also the Homing option checkbox in resume.

[image]

If not, please open the grbl-configuration form and read/refresh the configuration.

hk89, 2017-12-18:
Hi,

I guess I was not clear enough. Sorry, my bad. I meant that there was any option checkbox in the resume dialog window to resume from the last issued "work" position and it was next to the "redo homing" checkbox.

LASERGrbl would obtain the last work position before grbl went to a halt and issue a G92 command after grbl reset. Checkbox disappeared after v2.8.30 (I guess)

arkypita, 2017-12-18:
[image]

If you are talking about the checkbox "Home @ XY" she also appears only when necessary,

> LASERGrbl would obtain the last work position before grbl went to a halt 

Yes, LaserGRBL store the last known working position during file streaming, but the homing checkbox in resume dialog is shown only if a working offset was set (it is not zero). If the working offset was not set, it does not show any checkbox.

Are you sure you have a working offset settled when you have done this test? I have tried it right now and look working.

commit d550a961 (2017-12-03): fix #195

Better job resuming when a working offset was set