arkypita/LaserGRBL #199: laser notburnıng

77KIBRIS, 2017-12-04:
hı sınce that ı have downloaded ı had once prınted  1bıt pıc but that ıt sınce the upgrade no luck wıth laser prıntıng on anythıng ı have spoke wıth jay from jtech laser ıam usıng ( 3.8 laser ) he has told me to update to 1.1f grbl ı ahve done ıt but agaın no luck pls help ı am confused and stuch where ıam doıng wrong 
here ıs my confıguratıons

$0 | Step pulse time | 10
$1 | Step idle delay | 25
$2 | Step pulse invert | 0
$3 | Step direction invert | 3
$4 | Invert step enable pin | 0
$5 | Invert limit pins | 0
$6 | Invert probe pin | 0
$10 | Status report options | 3
$11 | Junction deviation | 0,020
$12 | Arc tolerance | 0,002
$13 | Report in inches | 0
$20 | Soft limits enable | 0
$21 | Hard limits enable | 0
$22 | Homing cycle enable | 0
$23 | Homing direction invert | 0
$24 | Homing locate feed rate | 25,000
$25 | Homing search seek rate | 0,000
$26 | Homing switch debounce delay | 250
$27 | Homing switch pull-off distance | 1,000
$30 | Maximum spindle speed | 255
$31 | Minimum spindle speed | 1
$32 | Laser-mode enable | 1
$100 | X-axis travel resolution | 250,000
$101 | Y-axis travel resolution | 250,000
$102 | Z-axis travel resolution | 250,000
$110 | X-axis maximum rate | 500,000
$111 | Y-axis maximum rate | 500,000
$112 | Z-axis maximum rate | 500,000
$120 | X-axis acceleration | 10,000
$121 | Y-axis acceleration | 10,000
$122 | Z-axis acceleration | 10,000
$130 | X-axis maximum travel | 1360,000
$131 | Y-axis maximum travel | 1290,000
$132 | Z-axis maximum travel | 68,000

thank you 
77kıbrıs

hk89, 2017-12-05:
Did you check your hardware twice? Did you check the parameters used in the JTech plugin like min and max power before exporting gcode? Do 'M03', 'M04' or 'S' appear in your exported gcode? I don't know from where to start.

Maybe try $32=0 before 'M03 S255' and then checking the output of the spindle enable pin with a multimeter.  Also attach your gcode here to have a look at it. This may provide a good start for providing solutions.

It has been a while since updates of LaserGRBL featured changes in the M03/M04 commands.

BTW, I thought I had a monitor issue when I have seen you using 'ı' instead of 'i'.  :D

77KIBRIS, 2017-12-05:
hı hk89 iam usıng turkish keyboard that why no dots :)))
ı have check the hardware all looks good to me thıs ıs the start of g code whıch dıt does appear to laser comes on and oss and power ıs m3 : 0 and m5 is : 255
check ıt pls 
[lasergrbl.docx](https://github.com/arkypita/LaserGRBL/files/1531339/lasergrbl.docx)
after thıs ı have press the play button and laser comes on and starts doıng the pıc but ıt wıll not burn
ı hope ıt does make sense and you can help me 
regards
77kıbvrıs

arkypita, 2017-12-05:
Please note that with the transition from grbl 0.8 to 0.9 (and therefore also 1.1) the laser pin and limit-z pin was exchanged. This was because the one used before did not have the pwm function. Look for grbl pwm on google and you should find what these pins are to be used.

However: M3 turn on laser, M5 turn off, on and off are managed by Grbl firmware inside arduino and there is **nothing in LaserGRBL** that can cause your problem (so this is not the right place to ask).

If you send M3 S255 command manually and the laser still remain off it mean you have connected the laser or ttl control to the wrong arduino pin. Check grbl documentation https://github.com/gnea/grbl/wiki/Connecting-Grbl

hk89, 2017-12-05:
Try $30=12000

arkypita, 2017-12-05:
> Try $30=12000

Why?

hk89, 2017-12-05:
I guess grbl 1.1 can use the 12000 as in grbl 0.9 Is it right?

77KIBRIS, 2017-12-05:
ı have tryed 12000 but not worked as arkypita when ı put the codes m03 s255 or others to come on or off does work but when ı run the gcode machıne works laser comes on  but doesnt burn on the wood?

arkypita, 2017-12-05:
$30=255 is ok
$31 should be 0

77KIBRIS, 2017-12-05:
I will check again thank you

iPhone'umdan gönderildi

arkypita, 2017-12-05:
@77KIBRIS respond to this questions:

1) do you have upgraded grbl? if YES from what version? 
2) does your laser driver support PWM/TTL modulation?
3) if YES at what arduino pin is connected TTL laser control?
4) with M3 S255 do you have laser turned on at full power?
5) with M3 S127 do you have laser turned on at half power?
6) with M3 S0 do you have laser turned off?

StuartB4, 2017-12-06:
Are you sure you have a working laser diode.
When they burn out they still emit light more like an LED light but they dont lase.

dungjk, 2017-12-06:
1. Make sure that you have correctly config grbl.
2. Check if your TTL module is connect correctly (if laser able to turn on/off while cutting then it's ok). Check grbl wiki for more information.
3. Make sure that your laser if *properly focused*. WEAR GOGGLES TO PROTECT YOUR EYES. Turn on the laser (Type M3 S<25% max spindle speed>, use M5 to turn laser off), turn the knob (CW or CCW) where the laser goes out to make the laser beam focus on material surface.

77KIBRIS, 2017-12-09:
thank you all for the help ıt works now only thıng ıs left to fınd outıs why when ı put above 250 on speed laser works but not burns

closed 2017-12-14 by arkypita