arkypita/LaserGRBL #200: Set Zero not permanent

TehseenHasan, 2017-12-06:
Setting work coordinates to zero at a specific position by pressing Set Zero button (icon : globe) to set zero position does not sets zero position permanently but it sets zero temporarily. And after restarting LaserGRBL software, if I go to zero position it does not go to last zero position...

arkypita, 2017-12-06:
The duty of persisting the position is not of LaserGRBL, but is a duty of grbl.
Grbl is ok in persisting working offset, and it only lost WCO if:
- power disconnect
- hardware reset (reset button of arduino)
- soft-reset (thunder button in arduino)

However after opening the connection LaserGRBL send a soft-reset in background, and this is why the coordinates are lost.

This behavior is not mandatory: I have inserted it for safety, to make sure to have all conditions resetted when you connect (speed override but also all modals like M3, M5 F, S etc).

If some users found it annoying / inconvenient I can add a configuration flag to prevent "auto soft-reset on connect".

TehseenHasan, 2017-12-06:
That will be very useful for me is it does not happens... Please change this setting or add option to Enable/Disable this setting....Thanks!

arkypita, 2017-12-06:
try https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.28

closed 2017-12-06 by arkypita

arkypita, 2017-12-08:
Dear @TehseenHasan 
Please note that LaserGRBL needs to know the actual grbl version (like v0.9b or 1.1e etc) to be able to self-configure its own feature based on the functions available in the various versions. 

If LaserGRBL does not reset grbl on connect, grbl does not send the welcome message like "Grbl 1.1e ['$' for help]" so LaserGRBL does not know the grbl version and issue can occur.