arkypita/LaserGRBL #203: Image Locating

bandguy335, 2017-12-07:
I like using this to run my laser engraver, but the only problem is being able to tell where the laser is actually going to start. Do you think there is a way to add like an image locating feature, or is there a way to add that to the custom buttons?

StuartB4, 2017-12-07:
It shows you where it's going to start with the blue broken line.

[image]

bandguy335, 2017-12-07:
I knew that but how do I where the laser is going to start on my machine bed.

bandguy335, 2017-12-07:
I guess I could have been a bit better at explaining my question. I know in the NEJE laser engravers they have a "Box Locating" button that shows you how big the engraving is going to be, and where exactly on the machine bed the engraving will happen, and that's what I was hoping to get a solution to,

hk89, 2017-12-07:
I got what you need. You need of locating the boundaries of a to-be engraved image. Solution is here in the wiki. Just add the 'M03' and 'M05' commands with your desired 'S' parameter. 
http://lasergrbl.com/en/usage/custom-buttons/

arkypita, 2017-12-14:
added some math ability in custom buttons
try [v2.8.31](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.31)

closed 2017-12-14 by arkypita