arkypita/LaserGRBL #207: Question Bypassing GBRL checks

marcwolf1960, 2017-12-11:
Hi Folks.
I have a GCode laser printer and I am modding the LaserGBRL code to talk to it.
One issue I have is trying to track how the system validated that it is talking to a GBRL engraver and bypass it.
I have been able to test things by using the ESP emulator and I am happy with the GCode that it produces. 

Any suggestions
Many thanks for your time
Dave

hk89, 2017-12-11:
Hi@marcwolf1960

You need a machine that runs grbl. LaserGRBL adapts itself on startup to a specific grbl version you have on a machine using commands like reseting grbl to know its version and sending in '$$' to know its configuration. Missing those will let LaserGRBL work as if in "Compatibility mode" where you can use it as a gcode generator only.

Hope that it helped

arkypita, 2017-12-11:
hi @marcwolf1960 

LaserGRBL talk with GRBL sending each gcode line and receiving response from grbl. Response for gcode commands could be "ok" or "error".

Meanwhile LaserGRBL continuously query for grbl status, sending the "query position" command each tot mSec. This is done in [ThreadTX()](https://github.com/arkypita/LaserGRBL/blob/0d99b038e5ff7e96114f9ed38feb5965fdbc9d97/LaserGRBL/GrblCore.cs#L969). Query position command cause grbl reply with real time status string.

The function [ThreadRX() in GrblCore.cs](https://github.com/arkypita/LaserGRBL/blob/0d99b038e5ff7e96114f9ed38feb5965fdbc9d97/LaserGRBL/GrblCore.cs#L1105) manage all the response from grbl. This includes both the receiving of ok/error and real time status response (and other messages).

The real time status string, sent by grbl, transport the current position AND the state of grbl controller.
This string look like `<Idle|WPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|WCO:0.000,0.000,0.000>`
This string contains different fields, separated by '|' char. The first field contains the state of grbl controller (in this case _Idle_).

[MacStatus enum](https://github.com/arkypita/LaserGRBL/blob/0d99b038e5ff7e96114f9ed38feb5965fdbc9d97/LaserGRBL/GrblCore.cs#L57) show all the possible state of grbl controller.

The whole LaserGRBL interface react to different MacStatus enabling/disabling buttons.
I.e. a file could be streamed only when _Idle_, the stop button is enabled only when _Run_ etc etc.

To fake/deceive LaserGRBL is connected you should call `SetStatus() `passing `MacStatus.Idle` when you connect adding `SetStatus(MacStatus.Idle);` ad the very last line of the `try block` in [OpenCom()](https://github.com/arkypita/LaserGRBL/blob/0d99b038e5ff7e96114f9ed38feb5965fdbc9d97/LaserGRBL/GrblCore.cs#L722)

arkypita, 2017-12-11:
You could fork your own version of LaserGRBL and modify it to work with your engraver.
I guess that your engraver is not managed by grbl... what kind of controller does it use?

marcwolf1960, 2017-12-11:
Many thanks Arkypita.  I have bee tracing the code but kept getting caught up with the various threads.

I have a TEVO Black Widow that uses Marlin 2.0 and a MKS 1.4 board. I have modded the firmware so I can use the normal GCode of M3,M4, and M5 to drive a laser, and a small CNC head.
Unfortunately I have found no CNC/Engraving programs that output GCode natively. All of them uses GBRL.
I tried changing my firmware on my MKS 1.4 and 'bricked' it. The  HEX I loaded had no bootstrap. However I was able to recover it with an ISP programmer.
If I can find some decemt GBRL firmware that I can mod then I don;t have an issue of swapping the HEX image around with XLoader.

Many thanks
Dave

arkypita, 2017-12-11:
You know that laserGRBL can export (File->Save Program) its own generated g-code file?

closed 2017-12-14 by arkypita