arkypita/LaserGRBL #209: Arduino e motori stepper

giampaolosacchetto, 2017-12-12:
Buongiorno

Recentemente ho realizzato un incisore laser con due motori stepper pilotati da LN298;
Le immagini da stampare vengono inviate da una applicazione android via bluetooth (HC05);

Tutto funziona ma ora voglio provare ad usare la libreria GRBL;

Ho installato su Pc il sw LaserGRBL e sull'ide arduino la libreria GRBL;

perciò il mio Sketch è il seguente:

`#include <coolant_control.h>
#include <defaults.h>
#include <eeprom.h>
#include <gcode.h>
#include <grblmain.h>
#include <limits.h>
#include <motion_control.h>
#include <nuts_bolts.h>
#include <planner.h>
#include <print.h>
#include <protocol.h>
#include <report.h>
#include <serial.h>
#include <settings.h>
#include <spindle_control.h>
#include <stepper.h>

#include <grblmain.h>

void setup(){
	startGrbl();
}

void loop(){}

`
Non capisco però a quali pin di arduino devo connettere i pin di LN298;

il driver del laser va al pin 11, ma i pin IN1 IN2 IN3 IN4 dell'LN298 ??? 

allego le foto dei pinOut di arduino e LN298
Ovviamente a me serve solo l'asse x e y  :

LN298
[https://drive.google.com/open?id=1uPCnbQo3viWpJUWC3ruLH7CEIRY_Yu8p
](url)

Arduino pin
[https://drive.google.com/open?id=18qmPTokX6K4N4LqyXpvmFkOCG7ikdpGv](url)

Giampaolo Sacchetto

Sono grato a chi mi aiuta a risolvere questo dilemma

arkypita, 2017-12-12:
Che io sappia grbl non può funzionare con gli L298, solo driver tipo poulu o allegro che si controllano via step/dir.

giampaolosacchetto, 2017-12-12:
Mi sa che hai ragione... è la soluzione migliore

Grazie mille del suggerimento

*Sacchetto GiampaoloVia Molini 50/e36051 Creazzo (Vicenza)cell. 349 1786031*

arkypita, 2017-12-12:
A metterci una scheda motori con driver poulu c'è solo da guadagnarci: microstepping, controllo correnti etc...

closed 2017-12-12 by arkypita

giampaolosacchetto, 2017-12-12:
Ho visto che pololu può erogare 1,5A mentre LN298 è più generoso (2A). Ma per i miei motori  va più che bene. Inoltre vedo che su amazon 5 pololu con dissipatore vengono poco meno di 12€