arkypita/LaserGRBL #210: Problem with engraving LaserGRBL preview

mibiosa, 2017-12-12:
Hi everyone,
I have a big problem with 2.8.30 version of LaserGRBL.
When I load any .nc code, in the preview image, all is red!! 
This is the issue:
[image]
And this is how it should be
[image]
LaserGRBL engrave correctly but I need the correct preview :-(
It gives me the same problem with all the images :-(
Does anyone have the same problem?
How I can fix that problem?
Please help meeeeee.
Thank you so much in advance.
Michelangelo

arkypita, 2017-12-12:
What software generate the .nc file?
Can you upload/post here any sample file?

mibiosa, 2017-12-12:
Thank you for your answer arkypita :-)
The software I'm using for generate the code is T2Laser.
But the problem is that with the previous version of LaserGRBL I never had that problem.
How I can upload the .nc file in this post?

mibiosa, 2017-12-12:
here it is:
[frontale scatola frandesco.zip](https://github.com/arkypita/LaserGRBL/files/1553453/frontale.scatola.frandesco.zip)

arkypita, 2017-12-12:
Is "Support PWM" enabled in your conf?

[image]

mibiosa, 2017-12-12:
Ok... fix!
It was not selected the option for PWM Support.... I enabled the option and now it works! :-)
[image]
Thank you so much Arkypita :-) Thank you very much for your hard and useful job :-)
Michelangelo

arkypita, 2017-12-12:
You are welcome, ciao!

mibiosa, 2017-12-12:
Ma scusa, sei italiano?

arkypita, 2017-12-12:
certo

mibiosa, 2017-12-12:
E non potevi dirlo subito ahahah io strippandomi a scrivere in inglese ahahahha
Comunque.... ti ho fatto una piccola donazione (non è tanto) per sostenere il tuo lavoro ;-)
Alla prossima.
Michelangelo

arkypita, 2017-12-12:
Grazie, molto gentile.

Anche se costa più fatica meglio mantenere l'inglese su queste pagine affinché possano essere d'aiuto un po' a tutti, inclusi i non italiani (che sono la maggior parte).

mibiosa, 2017-12-12:
Certo, assolutamente d'accordo... tanto la parte riguardante il problema l'abbiamo risolta in inglese ;-)
Grazie ancora e buona serata
Michelangelo

closed 2017-12-14 by arkypita