arkypita/LaserGRBL #211: change grbl configuration?

Evil-Nemesis, 2017-12-12:
hi there
how can I change 4 configs (for example $20~$23)  with coding?
for example in a button!
I use this way but this doesn't work well!

```
        private void Btn_Disable_Coolant_Click(object sender, EventArgs e)
        {
            Core.ExecuteCustombutton("M9");
            Core.ExecuteCustombutton("$20=1");
            Core.ExecuteCustombutton("$21=1");
            Core.ExecuteCustombutton("$22=1");
            Core.ExecuteCustombutton("$23=1");
        }
```

arkypita, 2017-12-12:
What do you mean, exactly, with "this doesn't work well"?

arkypita, 2017-12-12:
I recommend to use `Core.EnqueueCommand(new GrblCommand("gcode"))` instead of `Core.ExecuteCustombutton("gcode");` but ExecuteCustombutton is ok too.

Please note that setting configuration require eeprom writing and grbl has some issue with that:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#eeprom-issues

I suggest you to add some sleep using `System.Threading.Thread.Sleep(50);` between each configuration write.

Evil-Nemesis, 2017-12-13:
tnx for your answer my dear friend! ❤️  😍 
I really pray for you! 🙏 
yeah, but I found another way and it works fine too!

```
        private void Btn_Disable_Coolant_Click(object sender, EventArgs e)
        {
            List<GrblConf.GrblConfParam> write_conf;
            write_conf = Core.Configuration.ToList();
            foreach (GrblConf.GrblConfParam p in write_conf)
            {
                if (p.DollarNumber == "$20") p.Value = 1;
                if (p.DollarNumber == "$21") p.Value = 1;
                if (p.DollarNumber == "$22") p.Value = 1;
                if (p.DollarNumber == "$23") p.Value = 1;
            }
            Core.ExecuteCustombutton("M9");
            Core.WriteConfig(write_conf);
        }
```

Evil-Nemesis, 2017-12-13:
> What do you mean, exactly, with "this doesn't work well"?

Something happens as if the application is locked!
No G-code can be sent!
Anyway, thank you very very much! 💟 💕

closed 2017-12-13 by Evil-Nemesis