arkypita/LaserGRBL #212: Window Setting Request.

StuartB4, 2017-12-13:
Hi Arkypita.
Just wondered if it would be possible to add an On / Off option in the Settings to keep LaserGRBL's Window always on top. So it stays on top of any other windows.

arkypita, 2017-12-14:
I don't like always on top but ok... i can add it as a flag

StuartB4, 2017-12-14:
Ok, thanks very much.

arkypita, 2017-12-16:
I have tried to implement it quickly on mainform but it also require to change all the form to be "always on top" otherwise LaserGRBL dialog go under mainform and became not usable.
Since it require some wide changes in many form file I will delay this feature when I finish other works.

closed 2017-12-16 by arkypita

StuartB4, 2017-12-16:
Ok thats fine.
If it ends up being a problem for the rest of the program then it does not matter if you don't implement it.
It was just something I thought about, but may be I would be the only one to use it. I know a lot of people that don't like windows that you have to minimise to see other windows underneath.