arkypita/LaserGRBL #213: Show midle before Laser

JPVN, 2017-12-14:
Hi at all

Thanks for the programm!

I would like to set a custom botton how shows me the middle of the image to laser. If i can show the middle of the opened project it will be easier to pose the wood. Can you help me?

Thank you.

Best Regard JPVN

arkypita, 2017-12-14:
It is possible to move along the engraving border, not the middle.
http://lasergrbl.com/it/usage/custom-buttons/

In future i think to add some math interpretation ( + x - / ) to allow to compute different position other than left, right, top, bottom, width, height

StuartB4, 2017-12-14:
If you know the size of the image you are burning you can just use the lef/right/up/down and diagonal arrows to position the laser in the middle of the image on screen.
The laser will then move to the middle, then you can place your work piece under the laser and it will be over the middle.

arkypita, 2017-12-14:
try [v2.8.31](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.31)

closed 2017-12-14 by arkypita

JPVN, 2017-12-15:
It works! Very good. Thank you!

arkypita, 2017-12-15:
Good!