arkypita/LaserGRBL #214: PWM Modulation S options

mikepcurtis, 2017-12-14:
Hi, I have found the software very useful so far. Would it be possible to allow S ranges 0-1 for PWM modulation? currently it doesn't allow numbers <1.

Thanks

arkypita, 2017-12-14:
Why do you need this?

mikepcurtis, 2017-12-15:
PWM modulation in smoothieware requires an S value of 0-1 so to set a max value less than 100% an S value <1 is required

arkypita, 2017-12-16:
So your need is to use LaserGRBL as image-to-gcode generator for smoothie. As LaserGRBL is developed for grbl user i think i will keep the S input as integer type.
Maybe you can try using S0 to S99 and perform a "search and replace" inside the saved file to replace
"S" with "S." so S0 became S.0 and S99 became S.99

sns5400, 2017-12-18:
Smoothieboard expect for S a value between S0 -- S1 and activate the laser for every G1; G2 ; G3.
Switch off with G0 movements.
So no possible

closed 2017-12-21 by arkypita