arkypita/LaserGRBL #215: User-defined buttons

Hamewoker, 2017-12-15:
User-defined Butons,

Hello and thanks for the wonderful program.
 
What text do I have to write in the user-defined Butons to take over the feed rate as well as the steps that have been set before on a Z axis,
current solution is not really great with $ J = G01 F100 Z0.25
Thank you for your help
Sven
[image]
[image]

arkypita, 2017-12-15:
I'm not sure I understand your question.
Jogging specification could be found here: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Jogging

Hamewoker, 2017-12-15:
Yes misunderstanding,
Jogging is clear to me ...
Since in the program only directional keys left right forward back are available.
But I also have a Z axis where I then set the Focus over it I'm missing buttons for Z up and down.
currently solved via two custom buttons,
But so I can not use the defied F values and step values and so x times on it to print times to drive 10 mm or write these values over again.
And sorry Google translation

arkypita, 2017-12-15:
Now I understand.
I am currently work on a version able to compute expression in custom buttons.
I can add two new variable called jogfedd and jogspeed available to custom buttons code

Hamewoker, 2017-12-15:
Yes Perfect ,
I'm happy for that ...
Thank you in advance for the help
then I have something else, man can Vetoriezieren in the program but no Ai or SVG or other Vector files insert.
Would this be possible?

arkypita, 2017-12-15:
This is in dev roadmap but without promise on when i can implement it
https://github.com/arkypita/LaserGRBL/issues/64

Hamewoker, 2017-12-15:
No problem ...
and the list is still long.
I'm looking forward to the user buttons

arkypita, 2017-12-15:
version 2.8.32 is all you need :-)
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.32

closed 2017-12-15 by arkypita

arkypita, 2017-12-18:
You can write your custom button as `$J=G91 Z[jogstep] F[jogspeed]` to move up Z axis using values from speed/step slider and `$J=G91 Z[0-jogstep] F[jogspeed]` to move it down.

Hamewoker, 2017-12-18:
Thank you, and guilty I thought if shut you could not write in there anymore