arkypita/LaserGRBL #216: Wemos D1 mini and Arduino Uno connectivity

pjozsi, 2017-12-16:
Hi!
Unfortunately, you can not connect to the two devices. I have tried a lot with more arduino, GRBL 0.9 and GRBL1.1.
With USB connection, LaserGRBL works flawlessly.
For wifi connection at Arduino, only the RX Led flashes when it tries to connect. The serial monitor of IDE shows that data is coming back. Only TX Led is dark.
I did everything based on the description: Connect with Wifi.
Please help me, what's wrong, what am I doing wrong?

pjozsi, 2017-12-16:
session.log
2017.12.16. 0:31:46	Program     	------- LaserGRBL v2.8.30 START -------
2017.12.16. 0:31:49	SetStatus   	Machine status [Disconnected]
2017.12.16. 0:31:55	SetStatus   	Machine status [Connecting]
2017.12.16. 0:31:55	OpenCom     	Open ws://192.168.0.131:81/
2017.12.16. 0:31:55	SendImmediate	Send Immediate Command [0x18]
2017.12.16. 0:32:05	OpenCom     	Connection timeout!
2017.12.16. 0:32:05	CloseCom    	Close ws://192.168.0.131:81/ [CORE]
2017.12.16. 0:32:05	SetStatus   	Machine status [Disconnected]
2017.12.16. 0:32:12	SetStatus   	Machine status [Connecting]
2017.12.16. 0:32:12	OpenCom     	Open ws://192.168.0.131:81/
2017.12.16. 0:32:12	SendImmediate	Send Immediate Command [0x18]
2017.12.16. 0:32:22	OpenCom     	Connection timeout!
2017.12.16. 0:32:22	CloseCom    	Close ws://192.168.0.131:81/ [CORE]
2017.12.16. 0:32:22	SetStatus   	Machine status [Disconnected]
2017.12.16. 0:32:38	Program     	------------ PROGRAM STOP ------------

arkypita, 2017-12-16:
Hi @pjozsi 
Unfortunately only 8 LaserGRBL user in all the world use the connection via ESP8266 so I do not have much feedback about this feature. What I can say with certainty is that "it works", since I am one of these 8 users.

Is the IP adress 192.168.0.131 correct?
First do this test: Try to access with a browser to the page http://192.168.0.131/firmware and you should be prompt for username/password (admin/admin) 
If you see the firmware page, your wemos is programmed well.

Connection between wemos and arduino serial should be crossed (RX to TX and TX to RX)
If you attach serial debugger via USB cable or you open the COM port the communication between ESP and ARDUINO will be disturbed (of course) and cannot work.

Also remember that ESP modules use 3.3v and arduino use 5V. Somewere is suggested to connect via a 
voltage divider like this:

[image]

pjozsi, 2017-12-17:
Hi @arkypita!
Thank you for the advices! I managed to solve the problem.
Wemos requires a stronger voltage divider. I measured with oscilloscope RX pin of Wemos. The signal was high, never reached the low level.
Inner pull up resistor?
Now the voltage divider is 220 ohms and 470 ohms and works perfectly.
LaserGRBL is very cool!
Best regards

arkypita, 2017-12-18:
Good, Welcome to the WiFi VIP group!

I have bought one of this and it's all I need and all I want.

[image]

https://robotdyn.com/catalog/boards/uno_wifi_r3_atmega328p_esp8266_32mb_flash_usb_ttl_ch340g_micro_usb/

pjozsi, 2017-12-18:
This is very good, very universal. I'm interested  the development for this board of lasergrbl.
Good luck!

StuartB4, 2017-12-18:
Just a quick question.
Is that board going to work with GRBL.

I noticed that version 1.1 of GRBL uses nearly all of the space on the atmega328 chip.
Will there still be room for the other stuff the board needs to work.

arkypita, 2017-12-18:
I can confirm this board can work with grbl as I use it with grbl 1.1f
This board it is the hardware equivalent of an arduino one + ESP8266 on the same board connected together, eliminates the need to connect ESP externally and manages the conversion of 3.3-5V levels.

About the space it has the same space of any atmega328 board

arkypita, 2017-12-18:
@pjozsi 
About voltage divider, seem that Wemos has ESP RX/TX lines connected to USB-SERIAL chip via some 470ohm resistor. I don't know if CH340 lines are high or low impedence but this should be the reasons why you had to use lower ohmic values for voltage divider.

[image]

pjozsi, 2017-12-18:
Yes I think so too.
I have no experience with this board. I have one such, only ESP,
but I have not used it yet.
Yeah, I looked at the Wemos wiring diagram

https://www.aliexpress.com/item/ESP-12E-WeMos-D1-WiFi-uno-based-ESP8266-shield-for-arduino-Compatible/32821901911.html
[image]

pjozsi, 2017-12-18:
With the info, I have already ordered a UNO + WiFi R3 ATmega328P + ESP8266, thank you.

StuartB4, 2017-12-19:
> I can confirm this board can work with grbl as I use it with grbl 1.1f
This board it is the hardware equivalent of an arduino one + ESP8266 on the same board connected together, eliminates the need to connect ESP externally and manages the conversion of 3.3-5V levels.

> About the space it has the same space of any atmega328 board

Ok, thankyou very much.

closed 2017-12-19 by arkypita