arkypita/LaserGRBL #221: PC went to sleep while running a job

nhorvath, 2017-12-19:
Usually when I'm running LaserGRBL my PC stays awake but I came back and it had quit in the middle (and burned a hole clean through because the laser was on when it ran out of buffer).
Has it been a coincidence that some other program I was running was keeping the PC awake or is LaserGRBL supposed to? If it doesn't, would it be possible to tell windows to not let the PC sleep while running?
For now I will just change my power settings but I'd rather let it sleep when not running LaserGRBL.

Hamewoker, 2017-12-20:
You have to set the power options in Windows,
no matter of lasergrbl.
So that Windows does not go into sleep mode
Have this in general and have never had problems

nhorvath, 2017-12-20:
I am fully aware I can disable sleep in power settings but I know for a fact windows programs can tell the OS to prevent sleep while they are performing a task. I am just adding a feature request for that.

nhorvath, 2017-12-20:
It should be something like:
```// Job Started. Enable away mode and prevent the sleep idle time-out.
NativeMethods.SetThreadExecutionState(
    NativeMethods.ES_CONTINUOUS | 
    NativeMethods.ES_SYSTEM_REQUIRED | 
    NativeMethods.ES_AWAYMODE_REQUIRED);

// Wait until job is complete...
// Clear EXECUTION_STATE flags to disable away mode and allow the system
// to idle to sleep normally.
NativeMethods.SetThreadExecutionState(NativeMethods.ES_CONTINUOUS);```

Should be a simple change if you know where the methods that start / stop jobs are.

nhorvath, 2017-12-20:
MSDN Documentation on the above comment: https://msdn.microsoft.com/en-us/library/windows/desktop/aa373208(v=vs.85).aspx

arkypita, 2017-12-20:
good suggestion, I will implement this feature and i ask you to test
let me see at code

arkypita, 2017-12-20:
please try [v2.8.33](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.33) and let me know

nhorvath, 2017-12-21:
Thanks for the quick turn around. I'll try it next time I have something long to run.

closed 2017-12-21 by arkypita

frank42hh, 2021-06-08:
Sorry to hit on an old ticket, but has this been implemented and successfully tested? because I just had a run with v4.3.0 and my PC went into sleep after 20 minutes and killed my burn job.

arkypita, 2021-06-08:
LaserGRBL prevent OS from sleep using `SetThreadExecutionState(EXECUTION_STATE.ES_SYSTEM_REQUIRED);` [[note]](https://stackoverflow.com/questions/28751192/how-can-i-prevent-windows-os-from-entering-sleep-mode-net)

it was called every 20sec when in job.

arkypita, 2021-06-08:
I can do some testing but since v2.8.35 (4 years ago) no other user report your issue.

More possible you have USB sleeping, read lasergrbl faq https://lasergrbl.com/faq