arkypita/LaserGRBL #223: Laser starts out of the picture - GCode-Issue?

froetz, 2017-12-20:
Hello, 
I have an issue wit LaserGRBL (tried V2.8.22 and 2.8.30).

Here is a file, I made several weeks ago - this file works pretty fine:
[raclette-ask-code_working_nc.txt](https://github.com/arkypita/LaserGRBL/files/1576838/raclette-ask-code_working_nc.txt)

Today I tried to engrave a new picture, and my Laser keeps starting engraving at a wrong position and it only engraves a line a bit left and below the starting point, which is definded in the offset. I found out, that the first several lines in the gcode-file were different, and I tried to solve the problem. I used the old picture and created the gcode for it. The gcode has the same "bug or error" like the other file, but a few weeks ago, the same png-file was transferred into a good working gcode.
this is the picture:
[image]

This is the gcode that failed:
[raclette-ask-code_fail_nc.txt](https://github.com/arkypita/LaserGRBL/files/1576837/raclette-ask-code_fail_nc.txt)

I am unsing grbl 0.8c on an arduino uno. Old gcodes, I created weeks ago, are working fine. Only LaserGRBL creates a wrong gcode since 2 days. I started LAserGRBL as admin as well. I don't know exactly - but it could be, that the only thing that changed before installing 2.8.30 was the big win10-update. May be this information can help you figure out whats wrong?

Here is a video, that shows the problem: https://youtu.be/WeWoWy5xBz8
At 22 seconds, you will see where the laser is activated and doing only this movement

My GRBL-configuration:
$0=26.640 (x, step/mm)
$1=26.700 (y, step/mm)
$2=5.000 (z, step/mm)
$3=10 (step pulse, usec)
$4=50.000 (default feed, mm/min)
$5=500.000 (default seek, mm/min)
$6=160 (step port invert mask, int:10100000)
$7=30 (step idle delay, msec)
$8=75.000 (acceleration, mm/sec^2)
$9=0.050 (junction deviation, mm)
$10=0.100 (arc, mm/segment)
$11=25 (n-arc correction, int)
$12=3 (n-decimals, int)
$13=0 (report inches, bool)
$14=1 (auto start, bool)
$15=0 (invert step enable, bool)
$16=1 (hard limits, bool)
$17=1 (homing cycle, bool)
$18=122 (homing dir invert mask, int:01111010)
$19=200.000 (homing feed, mm/min)
$20=500.000 (homing seek, mm/min)
$21=50 (homing debounce, msec)
$22=1.000 (homing pull-off, mm)

Please tell my how I get LaserGRBL working again - It is the best software of its kind and it worked well -
 for me - till now.

arkypita, 2017-12-20:
For what I see in video "F0" (set speed at zero) cause G0 command to be refused and marked as error (red icon on the left). This is a different behaviour from grbl v1.1 I use for testing, that ignore F0 and accept the command.

F0 is an error, so I have to fix this bug. 
Meanwhile you can use an older version ([v2.8.16 is good](https://github.com/arkypita/LaserGRBL/releases?after=v2.8.17)) but I will release a fix as soon as possible... Stay tuned.

arkypita, 2017-12-20:
I also see another bug: names and description in grbl configuration form is wrong for older grbl version

[image]

arkypita, 2017-12-20:
please try [v2.8.33](https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.33) and let me know

froetz, 2017-12-21:
Thank you - I will triy, when I am home from work.

closed 2017-12-21 by arkypita

froetz, 2017-12-22:
Yeah - Thank you!
Now is seems to work!

commit 1aedcdc8 (2017-12-20): fix #223 #221