arkypita/LaserGRBL #225: Newest update laser stopped working

roko11, 2017-12-21:
Worked fine until update to 2.8.34.  Then had no laser output.Have grbl firmware 1.1e. Reinstalled 2.8.30 working fine again. All I did was uninstall 2.8.34 thru Windows 7 uninstall reinstall 2.8.30 and go.

arkypita, 2017-12-21:
@roko11 thanks for your fast feedback.
I have made some changes between 2.8.30 and 2.8.34 to fix some issue with GCode generated by raster import, maybe I made some error but I have tested with my engraver (grbl 1.1e too) without problem.

Can you explain me better how stopped working? What do you mean with "no laser output"?

StuartB4, 2017-12-21:
2.8.34 working fine here too. 
GRBL 1.1f

roko11, 2017-12-21:
What I mean is:
I have auto update on. When I opened the program it gave a message that the new version was available, I updated. I went to use the program and all looked normal. I opened a file I had used before and went thru the steps to compile and then I went to “burn” the file. No laser output.
I thought oh crap, my laser died. I had used the program with the old version the day before. So I shutdown and powered off. There is a sequence I use to start. I open the program, connect to the laser engraver then turn on the power. This prevents the laser from making a burn mark.
So I used a scrap of wood, disconnected the program, turned on the laser, connected and got my out of sequence burn mark. (Laser works)
I uninstalled the new version, downloaded and installed the version I had used previously. Works fine, no problem at all.
I have no issues with the old version. I normally don’t update  software just because a new version comes out, if I am not having issues with the older version.
That’s it in a nutshell. I will probably try the next new version, but if I have issues, I am happy to use the old.

From: arkypita 
Sent: Thursday, December 21, 2017 9:13 AM
To: arkypita/LaserGRBL 
Cc: roko11 ; Mention 
Subject: Re: [arkypita/LaserGRBL] Newest update laser stopped working (#225)

@roko11 thanks for your fast feedback.
I have made some changes between 2.8.30 and 2.8.34 to fix some issue with GCode generated by raster import, maybe I made some error but I have tested with my engraver (grbl 1.1e too) without problem.

Can you explain me better how stopped working? What do you mean with "no laser output"?

—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

arkypita, 2017-12-21:
Does the preview look like this? (gray instead of red or green)

[image]

arkypita, 2017-12-21:
I have fixed with v2.8.35
Sorry for the inconvenient, and thanks again for bug report. Let me know :-)

roko11, 2017-12-21:
In reply to earlier message I’m not sure what you meant by color, BUT new download of version 2.8.35 works fine with no issues. Thank You.

From: arkypita 
Sent: Thursday, December 21, 2017 1:15 PM
To: arkypita/LaserGRBL 
Cc: roko11 ; Mention 
Subject: Re: [arkypita/LaserGRBL] Newest update laser stopped working (#225)

I have fixed with v2.8.35
Sorry for the inconvenient, and thanks again for bug report. Let me know :-)

—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

closed 2017-12-21 by arkypita