arkypita/LaserGRBL #226: Track preview

mibiosa, 2017-12-21:
Will it be possible in the next versions to have a trace preview of the perimeter of the design? (like T2Laser)
Thank you :-)

vaicekas, 2017-12-22:
You can create your own button for this. There is even an example.
http://lasergrbl.com/en/usage/custom-buttons/

mibiosa, 2017-12-22:
Ok Thank you... I'll try :-)

Tihonuk85, 2017-12-23:
It works and it was so simple to make that button! Just amazing software. Thank you:)

mibiosa, 2017-12-23:
Could anyone recommend me a good manual (on internet) about the basic of the G-code (specific for DIY laser engraver machine?
I would like to try to create my own custom buttons for LaserGRBL.
Thank you in advance.
Michelangelo

hk89, 2017-12-23:
Here ya go http://reprap.org/wiki/G-code

mibiosa, 2017-12-23:
Thank you :-)
Another little question: I've tried to use the code at this link http://lasergrbl.com/en/usage/custom-buttons/ but it has a limitation because it traces only the borders of the image.
If I want to trace the full frame how can I do it?
I'll explain: I have an image that has blank borders around and I need to trace the full image to properly place the wooden piece in the engraver.
For example I have this image (I colored in pink the whole area I'm interesting in)
[image]
How can I modify the g-code to tell it to trace the full area borders? 
Thank you :-)
Michelangelo

hk89, 2017-12-23:
You need to crop that image in the software. You will find the editing tools under the preview image right after the import. Crop the image and select the area you are intersted in before creating the gcode.

mibiosa, 2017-12-23:
Ok. Thank you

vaicekas, 2017-12-24:
I use this way:
F5000
G0 X[left - 5] Y[bottom - 5]
F100
M3 S800
G1 X[right + 5]
G1 Y[top + 5]
G1 X[left - 5]
G1 Y[bottom - 5]
M5
G0 X0 Y0

mibiosa, 2017-12-24:
Doesn't exists the variable X maximum and Y maximum?
top, bottom, right and left refer to the vector image size but not to the entire image (with blank space around) :-(
For Example, the program T2Laser do it but it is not open source and I can't grab the code! (eheheheh)
When I load the image on it I can tell to preview trace the entire border of the image (with blank space around). 
I'm looking for a solution to copy this function to add my own button in LaserGRBL :-)

supertin, 2017-12-24:
+1 for the outline preview trace. While a custom button might work, I never would have thought of it.

Might be an idea to include a warning for first use if this gets added though... It wouldn't be nice to press it expecting a preview box, only to end up with a permanently burned box because your laser always runs at full power.

vaicekas, 2017-12-24:
Yes, I use these settings for the picture cut-out :)

arkypita, 2017-12-27:
> Doesn't exists the variable X maximum and Y maximum? top, bottom, right and left refer to the vector image size but not to the entire image (with blank space around) :-(

LaserGRBL compute left, right, top, etc variables directly from the generated GCode. This also allow to have this function working if you open an .nc /gcode file instead of a raster image.

Inside GCode there are no infos about the original image width and height.

mibiosa, 2017-12-27:
In fact, I open only .nc file.... the image that I've posted refer to a .nc code.
How can I extrapolate the data and pass it on to LaserGRBL to tell him the exact measurements I need?
The file .nc is this:
[DAVID right and left side.zip](https://github.com/arkypita/LaserGRBL/files/1589126/DAVID.right.and.left.side.zip)

arkypita, 2017-12-27:
If you want to trace this image frame with specular border you can use some math in a custom button

F100
G0 X0 Y0
G1 X[2*left+width]
G1 Y[2*top-height]
G1 X0
G1 Y0

arkypita, 2017-12-27:
pardon, i have edited. see now

arkypita, 2017-12-27:
Also... I see that .nc listing from T2Laser contains a header done using comments (round brackets)

[image]

This comments instruct T2Laser about some infos of the original file, this infos are good for T2Laser but they are not part of any standard, so i am not interested in parsing this data from the file and use them in any mode in LaserGRBL.

mibiosa, 2017-12-27:
Thank you... I'll try to see how it works :-)
I'll let you know :-)
Thanks again :-)
Michelangelo

mibiosa, 2017-12-27:
it would be interesting to use a variable to be able to assign the values ​​of X maximum and Y maximum but I don't know how to go and to take those two values

arkypita, 2017-12-27:
> I don't know how to go and to take those two values in the gcode
they simply do not exist in gcode

mibiosa, 2017-12-27:
I know that they do not exist... I have to find a solution to be able to take those two values.
If T2laser does it, surely there is a way ;-)
(to be specified that T2Laser does this by passing it any .nc file with the same structure of what I posted )

closed 2018-01-29 by arkypita