arkypita/LaserGRBL #227: Importing images for small engravings

e6910, 2017-12-22:
There was an entry about how images, when being rastered for small sizes got distorted. I'm engraving mostly small images.

You posted a workaround, that I would like to try programming a couple of buttons (one for changing the configuration and another to set it back to normal. I cannot find the instructions, Can you guide me to the post?

Marino

arkypita, 2017-12-23:
The hint is to scale the resolution setting in grbl
($100,101,102 if i remember well). This scale all drawing to 1/10 of the original size.

Now if you want to engrave something of 4x4 mm in LaserGrbl you design it 40x40 mm.

I.e. if your laser use 50 step to do one mm, you can set 5 step in conf.

arkypita, 2017-12-23:
I never try it personally but should work

e6910, 2017-12-25:
Generated gcode looks a helluva lot better, however, burning times for a 46x28 mm image rocketed from 6 minutes to 4.5hrs @14 lines...

e6910, 2017-12-25:
I though the degraded quality of small images was due to original image resolution, but that was not the case, as you said on another post.

In my case, I'm burning mostly small stuff...

arkypita, 2017-12-27:
Degraded quality of small images is because LaserGRBL scale (enlarge/reduce) the original image to the final image target. This is ok with enlargement but with reduction (expecially if small size) it produce a loss in detail in vectorization process.

Improvement in vectorization of small images is already in dev roadmap #64

arkypita, 2017-12-27:
> Generated gcode looks a helluva lot better, however, burning times for a 46x28 mm image rocketed from 6 minutes to 4.5hrs @14 lines...

This is because 14 lines mm now means 14 lines for each 1/10 of mm (140 lines/mm) i think

e6910, 2017-12-27:
Of course....

I think it would be best to wait.

Hope I could help...

Marino

closed 2018-02-27 by arkypita