arkypita/LaserGRBL #229: Shape Outline

DomStrict, 2017-12-24:
I'm sure it's possible but I can't find the functionality withing LaserGRBL, could someone let me know if burning the outline of a solid shape is available and if so how to set the software to do it.

Many Thanks

DomStrict, 2017-12-25:
I think it may be the vectorize setting, I hope to test this today and will post again.

StuartB4, 2017-12-25:
Yes, thats right.

Select your image.
[image]

In the Filling drop down menu select None. Then click Next.
[image]

Change any settings you need to here and click Create.
[image]

Run the program.
[image]

hk89, 2017-12-25:
I thought you meant a border of an engraved raster!

DomStrict, 2017-12-25:
Thanks Stuart

closed 2017-12-27 by arkypita