arkypita/LaserGRBL #230: Custom button - Low light intensity

DomStrict, 2017-12-25:
Does anyone have the code for a custom button to turn the laser on low light intensity so I can identify where the laser is positioned?

Or if the functionality already exists in the software please let me know.

Thanks in advance.

StuartB4, 2017-12-25:
Just type M3 in your custom button to switch laser on and another button with M5 to switch it off.
I think M3 S50 for example should switch the laser on with a power of 50.
I dont have a laser at the minute so I can't test it, need a new laser diode.

DomStrict, 2017-12-25:
Thanks for the info, I will test tomorrow and let you know how it goes.

abatinxs, 2017-12-25:
Take a new button and paste :
G01 F1000
M03 S1

DomStrict, 2017-12-26:
Fantastic:

The code that works for low laser intensity on for me is:

G01 F1000
M3 S10

To turn it off:

M5

Thanks everyone for your help.

DomStrict, 2017-12-26:
[image]

My New-look desktop

closed 2017-12-27 by arkypita