arkypita/LaserGRBL #232: Polish language

abatinxs, 2017-12-25:
Lot of Polish people using LaserGRBL. It will be nice if You introduce a Polish language to LaserGRBL. Thanks.

arkypita, 2017-12-27:
I will be proud to give support for Polish user with polish language.

Right now I'm working on a big restructuring of the code and of the user interface, so I suspended any translation jobs included new languages support. As soon as LaserGRBL version 3 will be out, I will resume translations job in the various languages, because they really need to be reviewed.

abatinxs, 2017-12-27:
arkypita very very thanks :) I`ll be waiting for this and waiting for dxf support.

closed 2018-02-27 by arkypita

my-pilot-life, 2019-03-14:
> I will be proud to give support for Polish user with polish language.
>
> Right now I'm working on a big restructuring of the code and of the user interface, so I suspended any translation jobs [...]

Helo
I can translate lasergrbl to Polish