arkypita/LaserGRBL #233: Framing Button - With Laser on low power (Resolved)

DomStrict, 2017-12-26:
So I wanted to see the area traced out by the low intensity laser.

The code that worked in my custom button is below.  You will need to select the appropriate power "S" code to suit your laser.  I have 2.5w module fitted at the moment.

I took the framin code from the website and added the laser on and laser off codes I used for my custom buttons.   I can now see exactly where my image will be burnt.

M3 S10
G0 X[left] Y[bottom] F1000
G1 X[right] F1000
G1 Y[top] F1000
G1 X[left] F1000
G1 Y[bottom] F1000
G0 X0 Y0 F1000
M5

StuartB4, 2017-12-26:
Good job Dom, glad it worked out ok.

Cant get that code to work on mine, just switches the laser on and then off again, but no movement.

arkypita, 2017-12-27:
You can set F1000 one a time at the beginning

F1000
M3 S10
G0 X[left] Y[bottom] 
G1 X[right]
G1 Y[top]
G1 X[left]
G1 Y[bottom]
G0 X0 Y0
M5

closed 2017-12-27 by arkypita