arkypita/LaserGRBL #234: G10 P1 L20 X0 Y0 Z0 Set work coordinates to Zero

TehseenHasan, 2017-12-26:
I was using G92 X0 Y0 Z0 g-code to **set zero position** (work coordinates) to engrave images on my laser machine... But with the G92 the new zero position is **not saved permanently** and when I restart LaserGRBL its previous zero location is lost... So **Now I use** this g-code **G10 P1 L20 X0 Y0 Z0** 
So please change your **Set Zero Point (G92)** button to G10 P1 L20 X0 Y0 Z0 so that last zero position saved permanently and we can work on the project even after restarting the machine on next day.
Thanks a lot.

DomStrict, 2017-12-26:
I found this too. Go to the Grbl option, then select settings , untick "Reset on Connect",  then your zero will be remembered when you log back in.

[image]

TehseenHasan, 2017-12-27:
Yes I know this option...But this is another issue...Actually (as far as I understand) GRBL firmware does not save work coordinates in EEPROM with G92 command. It saves work coordinates using G10 command.
According to GRBL Wiki : 

> G-code parameters store the coordinate offset values for G54-G59 work coordinates, G28/G30 pre-defined positions, G92 coordinate offset, tool length offsets, and probing (not officially, but we added here anyway). Most of these parameters are directly written to EEPROM anytime they are changed and are persistent. Meaning that they will remain the same, regardless of power-down, until they are explicitly changed. The non-persistent parameters, which will are not retained when reset or power-cycled, are G92, G43.1 tool length offsets, and the G38.2 probing data.

So set zero buttons in LaserGRBL should set work coordinates to EEPROM while now this button is setting tool offset rather than work coordinates. So this is the point I want to elaborate.

arkypita, 2017-12-27:
There is a problem in your G10 solution/suggestion: at hard reset (ie power on/off) the offset is persisted in EEPROM, but the current machine position is lost and restored to 0,0.  So after a restart you have issue because machine and world coordinate does not match. Just try!

1) G0 X100 Y100
2) G10 P1 L20 X0 Y0 Z0
3) hard reset
4) G0 X0 Y0

At point 4 the machine moves instead of remain in the last position.

TehseenHasan, 2017-12-27:
Yes you are right... Actually I have homing enabled and after restart I first home both axes then I send G0 X0 Y0 . Hence it works as I said before.

arkypita, 2017-12-27:
Ok, even if the solution is valid in your case, it can not be considered valid in any case (most user does not have homing ($H). This is why I prefer custom buttons instead of provide predefined buttons :-)

TehseenHasan, 2017-12-27:
Okay alright.. :-)

closed 2017-12-27 by arkypita