arkypita/LaserGRBL #235: latest download has malware

andyp60, 2017-12-27:
Trojan:Win32/Fuerboos.C!cl as identified by windows defender

arkypita, 2017-12-27:
1) LaserGRBL [is clean](https://www.virustotal.com/#/file/85744da1cff4570e1fba26fe5589f15ee52f0cdea8e3bf9daf8323b63aad84e2/detection)
2) Windows defender is worst antivirus ever
3) [False positive](https://www.google.it/search?client=opera&q=Fuerboos.C+windows+defender+false+positive&sourceid=opera&ie=UTF-8&oe=UTF-8) exist (always)

arkypita, 2017-12-27:
I have submitted latest install.exe to windows defender false positive service and they say "No malware detected" so please try to update your AV definition.

[image]

ermtl, 2018-01-15:
A good way to test each new release for false positives is to send it to https://virustotal.com as this will check against 68 antivirus engines

closed 2018-01-20 by arkypita

hayttle, 2018-10-17:
the same with me.

StuartB4, 2018-10-17:
No problems here with latest version, or any earlier version.
I've used the free version of Comodo Internet Security for years now and would highly recommend it.
It just works.