arkypita/LaserGRBL #236: Low laser, always or nothing

joelwedberg, 2017-12-28:
Hi!

Just updated my Firmware from Benbox´s old to GRBL 1.1 and running LaserGRBL 2.8.35 on the laser atm.

The problem I got is no mather what I do the laser does:

Running full power all the time over the passes(Min spindle speed 1000)
Running in half power but pauses passes he should (Min spindle speed 0)
Running off (adjusting max spindle speed)

I loaded the defaults_generic settings which is
#define DEFAULT_X_STEPS_PER_MM 250.0
  #define DEFAULT_Y_STEPS_PER_MM 250.0
  #define DEFAULT_Z_STEPS_PER_MM 250.0
  #define DEFAULT_X_MAX_RATE 500.0 // mm/min
  #define DEFAULT_Y_MAX_RATE 500.0 // mm/min
  #define DEFAULT_Z_MAX_RATE 500.0 // mm/min
  #define DEFAULT_X_ACCELERATION (10.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
  #define DEFAULT_Y_ACCELERATION (10.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
  #define DEFAULT_Z_ACCELERATION (10.0*60*60) // 10*60*60 mm/min^2 = 10 mm/sec^2
  #define DEFAULT_X_MAX_TRAVEL 200.0 // mm NOTE: Must be a positive value.
  #define DEFAULT_Y_MAX_TRAVEL 200.0 // mm NOTE: Must be a positive value.
  #define DEFAULT_Z_MAX_TRAVEL 200.0 // mm NOTE: Must be a positive value.
  #define DEFAULT_SPINDLE_RPM_MAX 1000.0 // rpm
  #define DEFAULT_SPINDLE_RPM_MIN 0.0 // rpm
  #define DEFAULT_STEP_PULSE_MICROSECONDS 10
  #define DEFAULT_STEPPING_INVERT_MASK 0
  #define DEFAULT_DIRECTION_INVERT_MASK 0
  #define DEFAULT_STEPPER_IDLE_LOCK_TIME 25 // msec (0-254, 255 keeps steppers enabled)
  #define DEFAULT_STATUS_REPORT_MASK 1 // MPos enabled
  #define DEFAULT_JUNCTION_DEVIATION 0.01 // mm
  #define DEFAULT_ARC_TOLERANCE 0.002 // mm
  #define DEFAULT_REPORT_INCHES 0 // false
  #define DEFAULT_INVERT_ST_ENABLE 0 // false
  #define DEFAULT_INVERT_LIMIT_PINS 0 // false
  #define DEFAULT_SOFT_LIMIT_ENABLE 0 // false
  #define DEFAULT_HARD_LIMIT_ENABLE 0  // false
  #define DEFAULT_INVERT_PROBE_PIN 0 // false
  #define DEFAULT_LASER_MODE 0 // false
  #define DEFAULT_HOMING_ENABLE 0  // false
  #define DEFAULT_HOMING_DIR_MASK 0 // move positive dir
  #define DEFAULT_HOMING_FEED_RATE 25.0 // mm/min
  #define DEFAULT_HOMING_SEEK_RATE 500.0 // mm/min
  #define DEFAULT_HOMING_DEBOUNCE_DELAY 250 // msec (0-65k)
  #define DEFAULT_HOMING_PULLOFF 1.0 // mm

And then changed laser to true

StuartB4, 2017-12-28:
Do you have PWM ticked in LaserGRBL settings menu.

joelwedberg, 2017-12-28:
Saw that! 
Thanks.

Now I got a serious scaling issue.
Sets picture to 50x50mm in paint, 50x50 in LaserGRBL but the laser just flies all the time. Have to manually stop him and reset.
Tested $13 $100 and $101 without any improvement more than 100 and 101 reduces the fault but seriously stops the speed

MXguy1000, 2018-02-13:
Speed is irrelevant, $100 and $101 MUST be set so that G0 X10 Y10 moves your location 10mm in X and Y from 0,0. Unless that is done you will always have "scaling" issues. You might try messing with the micro stepping settings on your motor drivers too.. But in the end 10 units in code must move your machine 10 units in reality, and that is done with $100 and $101..

closed 2018-02-27 by arkypita