arkypita/LaserGRBL #239: Some requests

jasoroony, 2018-01-01:
Just before opening the COM port, would it be possible to add the line of code "com.DtrEnable = true;" so Smoothieboards know LaserGRBL wants the connection message?

Also, not sure if it's my port of GRBL - but when I'm in Alarm state, I can't connect (seems to time out) but it could be because of the alarm message being sent on connect.

My laser isn't finished yet, so I can't tell if this is a good idea or not at the moment, but running your generated picture gcode on the bare electronics on my desk has long pauses on every "S0" command at the end of each line.  I can make them go away if I change it to "G1 X0 S0", relative mode means X doesn't actually move, but which tells GRBL not leave G1/2/3 mode and pause so much.

Thanks.

arkypita, 2018-01-29:
- Dtr managment added to dev roadmap
- question about alarm state not clear, explain better
- grbl has laser mode since v1.1 with laser mode enabled you have different managment of G1/2/3 with smoother movement and no pause

jasoroony, 2018-01-31:
Never mind the alarm state, I can't reproduce it anymore (at one stage I didn't have the emulated eeprom working on my 32 bit port and was seeing "error: 7" before the connection message - but even reproducing that isn't preventing a connection).  Not sure what happened there now.

Regarding laser mode, you're right - that generally works very well now..  However, if you set "Step idle delay" to 254 (the maximum) then just using S0 (and not something like G1 X0 S0) shows a quarter second delay at the end of each raster line when it leaves G1/2/3 mode and disables the steppers.

(Being on a 32 bit platform and having lots of flash space free, I've altered GRBL so the idle delay isn't a simple delay call in stepper.c when the stepper disable occurs, but is rather a timed idle loop check instead making it more asynchronous which also fixes the issue for me.  I.E. the steppers don't get disabled if a new gcode command comes in within 254ms.  But that's beside the point.)

Good work here by the way, this program makes using my laser much easier.

closed 2018-02-27 by arkypita

arkypita, 2018-04-10:
Version 2.8.43 add DTR managment
https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.43