arkypita/LaserGRBL #241: New please help. Image shifting

dtyers, 2018-01-03:
Hi my names Dan. I am new to laser engraving and am trying to get started engraving photos. I have run a couple of tests an got settings close to where I think the will work, but I have a problem with the image becoming shifted part way through running. I have included some pictures. One is of the test pattern and one is of two errors in gcode. I have had the happen in more than one occasion and with different images. Not sure what is going on or how to prevent the problem from reoccurring during a project in the future. Any help or advice would be greatly appreciated. Thanks
[image]
[image]

arkypita, 2018-01-03:
Look like https://github.com/arkypita/LaserGRBL/issues/24

arkypita, 2018-01-03:
Causes and solutions: http://lasergrbl.com/faq/

dtyers, 2018-01-04:
Thanks for the help

closed 2018-01-04 by dtyers