arkypita/LaserGRBL #246: Problem with autoupdate.

abatinxs, 2018-01-09:
When I run LaserGRBL (now 2.8.35) the window appears with info - new version available 2.8.8. When download is ends, I see the info - automatic update failed. Please manually download the new version from lasergrbl site. This is problem on my computer, the server or new update lasergrbl? And last - I don`t see 2.8.8 version for download :)

arkypita, 2018-01-09:
Are you running it on windows XP?

abatinxs, 2018-01-09:
I running on Windows 7 64bit (Dell E4310)

arkypita, 2018-01-09:
This issue is very strange.

v2.8.35 is actually the latest (most updated) version. LaserGRBL detect the latest version checking the content of the url: https://api.github.com/repos/arkypita/LaserGRBL/releases/latest

[image]

The content of "tag_name" is used as latest version. Actually it show "tag_name": "v2.8.35"

V2.8.8 is very old. I have no idea of why your installation found this version as latest.
What do you see if you open https://api.github.com/repos/arkypita/LaserGRBL/releases/latest in your browser?

abatinxs, 2018-01-09:
When I open this link I see :
"html_url": "https://github.com/arkypita/LaserGRBL/releases/tag/v2.8.35",
  "id": 8999345,
  "tag_name": "v2.8.35",

First printscreen:
https://i.imgur.com/7aW4ADU.jpg

And second printscreen with what I see when open Your link
https://i.imgur.com/kWy2BrF.jpg

Github not support to view a photos in this window. You must copy the link and paste in Your browser.

arkypita, 2018-01-09:
Please note that you are using an old v2.8.8 version. LaserGRBL simply warnibg you that latest version is 2.8.35.

The auto update procedure is changed and it is not compatibile with so old version, so auto update fail.

Simply download and install manually the latest v2.8.35

arkypita, 2018-01-09:
Ps github support alt+stamp or win+stamp (take screenshot) and ctrl+v to paste in a comment, without need use any external service

closed 2018-01-10 by arkypita