arkypita/LaserGRBL #248: LaserRGBL id disconected to Arduino UNO or Arduino Mega

Yanabalts, 2018-01-14:
[1sessionlog.txt](https://github.com/arkypita/LaserGRBL/files/1629821/1sessionlog.txt)
Hello Dear @arkypita!

I try do connect LaserRGBD to Arduino Mega2560

What I see:
Disconnected>Connecting>Diconnected

It looks like that https://pastenow.ru/2HUJB

I Also add session log file in attachment.

Could you kindly tell me what I do wrong. I would really love to use you program.

StuartB4, 2018-01-15:
Change the BAUD Rate to 115200 in the drop down menu. Then click Connect.
[image]

arkypita, 2018-01-15:
@Yanabalts What version of grbl have you flashed on your arduino?

Yanabalts, 2018-01-15:
Thank you very much! Right after I added RGBL to my Arduino It was connected.
Thank you!
Could you tell me also where can I find PINS for LaserGRBL working with Arduino Uno. What pins are for laser?

arkypita, 2018-01-16:
Why not [google ](http://lmgtfy.com/?q=grbl+pinout) for that?

lucasssimao, 2018-01-17:
Hello i having the same problem but i dontknow how to flash the arduino, i've tried to change the BAUD rate but doesn't work, could you helpme pls ? tks

arkypita, 2018-01-29:
@lucasssimao 

If you search on google you can find what you want.
https://github.com/gnea/grbl/wiki/Flashing-Grbl-to-an-Arduino

Please note:
- grbl is not provided my me
- grbl official site: https://github.com/gnea/grbl/
- flashing a different firmware to your eleksmaker can break your engraver, if you don't know how to do that safely ask for support to http://eleksforum.eleksmaker.com

closed 2018-02-27 by arkypita