arkypita/LaserGRBL #249: Customized Buttons

Master-Disaster, 2018-01-15:
First of all: Great programm. Thanks for that.
I want to build a big laser engraver ( 3,5m x 3,5m). 
As i want to do some files first, I recognized, that my laptop is not powerfull enough to calculate big files.
In this case, i need to transfer my system to another laptop.
Is there a possibility to see and copy a file in the computer for my customized buttons?

It would be much easier to copy a file than to create new buttons on new computer.

Kind regards in advance

arkypita, 2018-01-15:
3.5m x 3.5m is very huge! I think it will be the biggest frame LaserGRBL will work with.
I think you have read this topic: https://github.com/arkypita/LaserGRBL/issues/139

I have tested LaserGRBL with big sizes, but I want to warn you that an image of 3.5m x 3.5m it is very very big also for LaserGRBL. This kind of image processed @ 10lines/mm require that LaserGRBL generate a bitmap of 35000*35000 pixels, that in memory require about 5GB.

Since LaserGRBL should load the image in memory more than one time (i.e. during transformation like from color to grayscale the image exist in memory both colored and grayscaled) the managment of this size of image will require easly 15-20 GB of RAM otherwise the swap on disk will be used and the process will become very slow.

arkypita, 2018-01-15:
However I have not understand your question.
What do you want to do with custom button?

Master-Disaster, 2018-01-15:
I created some custom buttons. But for this memory reasons you explained I
need to change my laptop. The old laptop goes in error while creating file.

I want to copy.my exists g buttons to the new laptop. But I don't know how
to transfer the buttons.

Kind regards

Klaus

arkypita, 2018-01-15:
C:\Users\YOUR NAME\AppData\Roaming\LaserGRBL\CustomButtons.bin is your friend.

Master-Disaster, 2018-01-15:
Thank you.
Now i found the file.

I have a second problem today.
Should i open a new issue point or could you help me by email?

I tried to make a wooden plate with a pizza symbol.
The result is crazy.
I attached a picture from screen, result and session.log.

Kind regards in advance

Klaus

> 
> C:\Users\YOUR NAME\AppData\Roaming\LaserGRBL\CustomButtons.bin is your friend.

1/14/2018 2:45:56 PM	OpenFile    	Open C:\Users\Sontaya\Pictures\Laser-Engraver\Brotzeit.jpg
1/14/2018 7:20:45 PM	SendImmediate	System.UnauthorizedAccessException: Access to the port 'COM1' is denied.

1/14/2018 7:20:45 PM	SendImmediate	   at System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
1/14/2018 7:20:45 PM	SendImmediate	   at System.IO.Ports.SerialStream.EndWrite(IAsyncResult asyncResult)
1/14/2018 7:20:45 PM	SendImmediate	   at System.IO.Ports.SerialStream.Write(Byte[] array, Int32 offset, Int32 count, Int32 timeout)
1/14/2018 7:20:45 PM	SendImmediate	   at System.IO.Ports.SerialPort.Write(Byte[] buffer, Int32 offset, Int32 count)
1/14/2018 7:20:45 PM	SendImmediate	   at LaserGRBL.ComWrapper.UsbSerial.Write(Byte b)
1/14/2018 7:20:45 PM	SendImmediate	   at LaserGRBL.GrblCore.SendImmediate(Byte b, Boolean mute)
1/14/2018 7:20:45 PM	SetStatus   	Machine status [Disconnected]
1/15/2018 10:09:43 AM	Program     	------- LaserGRBL v2.8.35 START -------
1/15/2018 10:09:46 AM	SetStatus   	Machine status [Disconnected]
1/15/2018 10:19:54 AM	Program     	------------ PROGRAM STOP ------------

1/15/2018 10:20:21 AM	Program     	------- LaserGRBL v2.8.35 START -------
1/15/2018 10:20:24 AM	SetStatus   	Machine status [Disconnected]
1/15/2018 10:29:40 AM	SetStatus   	Machine status [Connecting]
1/15/2018 10:29:40 AM	OpenCom     	Open COM1 @ 115200 baud
1/15/2018 10:29:40 AM	SendImmediate	Send Immediate Command [0x18]
1/15/2018 10:29:40 AM	VersionInfo 	Detected Grbl v0.9i
1/15/2018 10:29:41 AM	SetStatus   	Machine status [Idle]
1/15/2018 10:29:49 AM	OpenFile    	Open C:\Users\Sontaya\Pictures\Laser-Engraver\Brotzeit.jpg
1/15/2018 10:30:06 AM	SetStatus   	Machine status [Run]
1/15/2018 10:30:21 AM	SetStatus   	Machine status [Idle]
1/15/2018 10:30:27 AM	SetStatus   	Machine status [Run]
1/15/2018 10:30:42 AM	SetStatus   	Machine status [Idle]
1/15/2018 10:30:56 AM	SetStatus   	Machine status [Run]
1/15/2018 10:31:07 AM	SetStatus   	Machine status [Idle]
1/15/2018 10:31:11 AM	SetStatus   	Machine status [Run]
1/15/2018 10:31:26 AM	SetStatus   	Machine status [Idle]
1/15/2018 10:33:45 AM	SetStatus   	Machine status [Run]
1/15/2018 10:33:45 AM	SetStatus   	Machine status [Idle]
1/15/2018 10:33:50 AM	SetStatus   	Machine status [Run]
1/15/2018 10:34:16 AM	SetStatus   	Machine status [Idle]
1/15/2018 10:35:34 AM	OpenFile    	Open C:\Users\Sontaya\Pictures\Laser-Engraver\can-stock-photo_csp12609078.jpg

1/15/2018 10:35:56 AM	CloseCom    	Close COM1 [USER]
1/15/2018 10:35:56 AM	SetStatus   	Machine status [Disconnected]
1/15/2018 10:35:56 AM	Program     	------------ PROGRAM STOP ------------

1/15/2018 10:36:04 AM	Program     	------- LaserGRBL v2.8.35 START -------
1/15/2018 10:36:07 AM	SetStatus   	Machine status [Disconnected]
1/15/2018 11:32:34 AM	OpenFile    	Open C:\Users\Sontaya\Pictures\Laser-Engraver\Pizza_Schnitte_1.jpg
1/15/2018 11:33:57 AM	SetStatus   	Machine status [Connecting]
1/15/2018 11:33:57 AM	OpenCom     	Open COM1 @ 115200 baud
1/15/2018 11:33:57 AM	SendImmediate	Send Immediate Command [0x18]
1/15/2018 11:33:58 AM	VersionInfo 	Detected Grbl v0.9i
1/15/2018 11:33:58 AM	SetStatus   	Machine status [Idle]
1/15/2018 11:34:02 AM	SetStatus   	Machine status [Run]
1/15/2018 11:34:30 AM	SetStatus   	Machine status [Idle]
1/15/2018 11:34:34 AM	SetStatus   	Machine status [Run]
1/15/2018 11:35:50 AM	SetStatus   	Machine status [Idle]
1/15/2018 11:36:41 AM	EnqueueProgram	Running program, 33484 lines
1/15/2018 11:36:41 AM	SetStatus   	Machine status [Run]
1/15/2018 12:04:55 PM	RealTimeStatus	Ex on [Run,MPos:54.938,64.938,0.000,00,WPos:56.100,63.775,0.000] message
1/15/2018 12:04:55 PM	RealTimeStatus	System.ArgumentOutOfRangeException: startIndex cannot be larger than length of string.
1/15/2018 12:04:55 PM	RealTimeStatus	Parameter name: startIndex
1/15/2018 12:04:55 PM	RealTimeStatus	   at System.String.InternalSubStringWithChecks(Int32 startIndex, Int32 length, Boolean fAlwaysCopy)
1/15/2018 12:04:55 PM	RealTimeStatus	   at LaserGRBL.GrblCore.ManageRealTimeStatus(String rline)
1/15/2018 12:41:06 PM	RealTimeStatus	Ex on [Run,MPos:109.200,,93.438,0.000,WPos:108.025,93.438,0.000] message
1/15/2018 12:41:06 PM	RealTimeStatus	System.FormatException: Input string was not in a correct format.
1/15/2018 12:41:06 PM	RealTimeStatus	   at System.Number.StringToNumber(String str, NumberStyles options, NumberBuffer& number, NumberFormatInfo info, Boolean parseDecimal)
1/15/2018 12:41:06 PM	RealTimeStatus	   at System.Number.ParseSingle(String value, NumberStyles options, NumberFormatInfo numfmt)
1/15/2018 12:41:06 PM	RealTimeStatus	   at System.Single.Parse(String s, NumberStyles style, NumberFormatInfo info)
1/15/2018 12:41:06 PM	RealTimeStatus	   at LaserGRBL.GrblCore.ManageRealTimeStatus(String rline)

arkypita, 2018-01-15:
> I have a second problem today. Should i open a new issue point or could you help me by email?

Each reply you send to email is received by github and interpreted as a reply to the issue.
See #249

arkypita, 2018-01-15:
Image attachment sent by mail was dropped by github.

Master-Disaster, 2018-01-18:
[image]

Master-Disaster, 2018-01-18:
There is a diagonal shift in the result

arkypita, 2018-01-18:
You have electrical poor quality arduino clones or electrical noise on serial line.

http://lasergrbl.com/faq/
#24
#55
#81
#118
#241

Master-Disaster, 2018-01-24:
Hi
I made some more trials with different results.

First i changed all cable connections, but the result became worst.
1. Pizza_2
[image]
2. Pizza_horizontal
[image]
3. Pizza_diagonal
[image]

I recognized, that with every test the result become worst more than before.
After this i deinstalled your program and installed it new.

Now everything works fine.
4.Pizza_final
[image]

5.Laser_38x26
[image]

arkypita, 2018-01-29:
The reinstall fix your problem sound strange for me, but if it work... good!
However, if you still have problem in the future follow guidelines at the end of http://lasergrbl.com/faq/

closed 2018-01-29 by arkypita