arkypita/LaserGRBL #250: New parameters for better quality engravings

ermtl, 2018-01-15:
When using LaserGRBL with a cheap (wobbly) laser engraver, the border must be done at low speed to prevent it from being awfully distorted while the filling can be done much faster (even more so when the laser is more powerful). This is addressed with the different "filling speed" and "Border Speed" parameters.
However, when the speed difference is too big, the borders are over-burnt as the S MAX setting applies to both filling & border. Adding a different set of settings for these 2 operations would make sense.

Also, as a high acceleration (GRBL parameters $120 and $121) makes things much worse, being able to scale the value for both X & Y when drawing the borders would result in a much more precise border without slowing down the filling operation.

here is an example of what the modified parameter screen could look like:
[image]

arkypita, 2018-01-29:
Is a good suggestion but make the interface more complicated. My goal is to keep the sw as simple as possible, for a newbie all this option can confuse.

3ditguy, 2018-01-29:
I too would like this feature, could you have an "advanced" toggle button
to show/hide some additional fields?

closed 2018-02-27 by arkypita