arkypita/LaserGRBL #252: losing 0

rkontz, 2018-01-21:
I put a new board on my laser and now its start cutting fine but have way through the program it changes it 0 point 
[image]
[image]

[image]

arkypita, 2018-01-21:
You have poor quality arduino clone or electrical noise on serial line.

http://lasergrbl.com/faq/
#24
#55
#81
#118
#241

rkontz, 2018-01-22:
Thank you . I had mounted the board on the laser . So i took it off an now seens to be running  go . It been running for 6 hour with no trouble

Happy Connecting. Sent from my Sprint Samsung Galaxy S® 5 Sport

http://lasergrbl.com/faq/
#24<https://github.com/arkypita/LaserGRBL/issues/24>
#55<https://github.com/arkypita/LaserGRBL/issues/55>
#81<https://github.com/arkypita/LaserGRBL/issues/81>
#118<https://github.com/arkypita/LaserGRBL/issues/118>
#241<https://github.com/arkypita/LaserGRBL/issues/241>

closed 2018-01-24 by arkypita