arkypita/LaserGRBL #253: Enhancement, can we have the "S" current value on screen where the x-y values are

sinfocomp, 2018-01-22:
it would be nice to see the current laser powe on screen

arkypita, 2018-01-24:
added to roadmap
must check if it is possible (aka reported by grbl realtime status message)

closed 2018-01-24 by arkypita

nico48, 2018-02-12:
How do you install this program?  I have linux suze, and i cannot see installation program...where did you put that? ( i have googled all morning to figure it out myself....)

arkypita, 2018-02-12:
First @nico48 please do not post haphazard (at random). This thread is talking about enhancement of showing S value on screen. Your question is not related to this, so open a different issue.

Second LaserGRBL is Windows only.

nico48, 2018-02-12:
ok my bad, and the last remark explains a lot thanks !!

arkypita, 2018-02-12:
@nico you are welcome
For run LaserGRBL on linux read this post: https://github.com/arkypita/LaserGRBL/issues/5