arkypita/LaserGRBL #254: Laser to weak?

beckfly, 2018-01-23:
Hello,I'm new here and I have a question about my new Laser engraving mashine.
I have bought an EleksLaser Pro with the size of A3.The Laserpower is 0.5W.
I have assembled everything yesterday and it runs perfect with the software lasergbrl.
Today I have make the first tests.  I have tryed to focused the lence of the laser.Nothings happen.
When the Laser is quite a wihle on a Point,the Laser starts slowly to make a point of the cardboard.
So maybe the laser is to weak or what can I do or change?
Maybe sombody can help?
Michael

abatinxs, 2018-01-24:
Replace the power in LaserGRBL with S255 to S1000. I had the same witch 3,5W laser.Do this in window with speed (F), commands on  off laser M3/M5 size object and offset.

beckfly, 2018-01-25:
Hello,
thank you for your help.I have Chance this Settings And now everything Works.
Now i can Start to learn the possibilitis of my machine.
Kind Regards!
Michael

closed 2018-01-29 by arkypita