arkypita/LaserGRBL #257: Quality (lines/mm) - Less Than 1 ?

MarlynAnderson, 2018-01-27:
Not an issue - just a wish.  OK - I'm weird - I need less resolution. I'm melting foam. The surface is above focal point so horizontal scan from blue laser diode melts a 1.5mm 'tall' swath.  The goal is to get the 100 sq.px  image to be 150mm sq. - made with 100 scan lines (and laser intensity adjusted 100 times per line).  The Quality value won't go below 1.000 line/mm (I'd like 0.75). My stepper res is 19.685 steps/mm. I could probably fake it out with $100,$101=30, but I might not remember to change it back before quiting for the day (so it doesn't mess up the next thing I do next week) - so I'm wishing for  '1 line /1.5mm' to work.
Could it be as simple as 'Remove the '<1' error-check and let the math do it's thing.' ?
OR... is there an alternate setting I'm ignorant of?
BTW - can't wait to make $$ using this so I can DONATE! :-)

arkypita, 2018-01-29:
Hi @MarlynAnderson 
I understand your needs. Unluckily the code of LaserGRBL is developed using an integer number as data type for the value of resolution. This is spread in the code and i must look in deep about that.
When i have some time i get a look to see of it is easy to implement what you ask for.

For now, i add to dev roadmap #64

MarlynAnderson, 2018-01-30:
Thanks for considering this - but as I represent a fraction of 1% of present and future users - and even then it's mostly about the fear of a future boo-boo due to 'lack-of-attention-to-detail', I have to say, "Feel free to move it way to the bottom. " :-) 
PS.  Too bad the GRBL startup blocks ($N0,$N1) don't accept '$' things.  Setting $100 and $101 from there while using your nice 'GRBL config.' dialog for the needs of the moment would render this issue moot. 
PSS. In the meantime... Perhaps changing the scroll box number format to "#", rather than "#.###"  would be a thing to do - keep people from getting the impression that fractions can be considered?
Thanks again - wonderful job you've done - it works really smoothly once feed/power rates are 'dialed in'!

closed 2018-02-27 by arkypita

arkypita, 2021-10-14:
Added in v4.6.0