arkypita/LaserGRBL #259: QUESTION, is there a stop or abort function other than a reset?

sinfocomp, 2018-01-30:
hello is there an abort or stop function other than the reset button?
thanks

arkypita, 2018-01-31:
No, there isn't.
But what do you mean with abort: simply stop sending gcode or what?

sinfocomp, 2018-02-05:
If I pause the job and want to restart I need to reset, and if I reset I loose position and need to home again, it would be nice to be able to abort a job without loosing position, also it would be very helpfull to be able to run gcode from a specific line of code, thanks

arkypita, 2018-02-08:
understand. can be added in a future version

arkypita, 2018-02-08:
I think that could be some problem that can happen if job is aborted without reset grbl.

If you pause job (pause is executed immediately) some command could remain in grbl buffer (arduino side) waiting for job resume command.

If I provide an abort button, implemented like "discard any command not yet sent from LaserGRBL and put in"

closed 2018-02-08 by arkypita