arkypita/LaserGRBL #261: laser jump to other point and continue burn

soulshot202, 2018-01-31:
Hi, im new and have a isue. i try to burn small images (60x30mm) but laser after few mins from start burning missing last point and continues burn picture from other point,i see problem in photos. my firmware are 0.9j, my laser settings also are in picture. i try with 1.1 firmware,then laser after few mins burns one line,but not continue picture from other point. i try put lower rezolution pictures into program, but always same problem.also i try burn image 110x55mm, and also same problem with jumping lines from other point... i use laser gbrl version 2.8.35, and CH340 (08/08/2014, 3.4.2014.08) driver. Anyone can help me?
[image]
[image]

[image]

soulshot202, 2018-01-31:
also i try with diferent engrave speed 1000-5000, and try with linetoline and dithering modes...but always same problem..

arkypita, 2018-01-31:
You have poor quality arduino clone or electrical noise on serial line.
LaserGRBL use a ultra-fast streaming protocol between PC and arduino but cheap chinese clone fail receiving data. This could be also caused by bad usb cable or motor noise.

http://lasergrbl.com/faq/ (read all and look at the end of the page for a sw solution).
#252 
#24
#55
#81
#118
#241

closed 2018-01-31 by arkypita